bambulab/BambuStudio #1245: 3mf import loses data

yahbluez, 2023-02-04:
**Describe the bug**

Loading/importing a 3mf file made with any other application **will lost any data** but the model data.

**3mf File for This Bug**

https://files.printables.com/media/prints/251587/stls/2716404_07447c22-0d7f-4a66-9385-ced1f1ffe233/firstlayercalibration-firstlayer.3mf#_ga=2.54709357.734743191.1674145992-1410400879.1648631397

**To Reproduce**
1. Import / open the linked (or any other not made by bamdustudio) 3mf file
2. see how all configuration data is lost

**Expected behavior**

Import at least all the configuration data from settings bambustudio is using.
For example the "infill angel" or the infill pattern or the extrusion wide or etc.

**Screenshots**

Not necessary

**Desktop (please complete the following information):**

Happens with all versions of bambustudio

lanewei120, 2023-02-05:
the configurations are not in 3mf standard
and they are different between different vendors

we can not support other vendor's configurations

yahbluez, 2023-02-05:
This is not a vendor configuration. 
It is user data which are deleted while importing into bambustudio.

**Things like bottom layer infill angel are not vendor specific.**
This is important user data.

This is a real bug in your software. 

Your software destroys hours of work, 
you should get aware of that bug  and not deny it. 

Where is the problem to import well known attributes?
Many other slicers can do that.

lanewei120, 2023-02-06:
> 

Hello

the 3mf you provided are saved by prusaslicer,
the configuration are saved in a prusa specific format
most likely only Prusa&&SuperSlicer(they use the same 3mf format) can only the file with the configuration kept

althrough "bottom layer infill angel" is a common configuration,
it is just one configuration included in all configurations,
which is saved in the whole configuration file too

we can not load the vendor configuration file because there are lots of other configurations different

BTW althrough we are derived from PrusaSlicer, the configuration format has been changed
and the configurations have big differences now

thanks

closed 2023-03-02 by SaltWei