bambulab/BambuStudio #1246: Need PID tuning

npoulos, 2023-02-05:
**Describe the bug**
I’ve found when I print with a bed temp of 60 or higher, my prints have this spiral artefact. I’ve done all the regular maintenance and cannot get it to go away. 

Prints that are 30-35 ped temp do not have the artefact. 

It seems to me that the bed is periodically taking power from the hot end. 

**3mf File for This Bug**
using bambulab standard profile for 0.2 height and 0.4 nozzle 

**Expected behavior**
Consistent extrusion

**Screenshots**
[image]

bottom half of cylinder is with a bed temp of 35 and top half a bed temp of 75. You can see clear artefact difference. 

**Desktop (please complete the following information):**
macOS

npoulos, 2023-02-05:
[image]

[image]

theres  a very clear line distinction on the print when the bed is instructed to change temperature by the added M140 codes at the corresponding heights. 

The “deepest” impression is when the bed changes from essentially off M140 S0 at the bottom to on M140 S40. 

It’s very clear that the heat bee causes strange artefacts in the print as it tries to keep itself hot. 

Some sort of PID tuning would be good to be able to stop these artefacts.

npoulos, 2023-02-05:
[image]

this test I started the print with a bed temp of 75 and set it to 0 on layer 2. You can see none of the screw artefacts as the hot bed was essentially off. 

This shows it’s not the temperature causing a problem but rather the hardware trying to keep the temperature stable/heat up.

npoulos, 2023-02-05:
[TempTestCylinder.gcode.3mf.zip](https://github.com/bambulab/BambuStudio/files/10613746/TempTestCylinder.gcode.3mf.zip)

Here is a test file that should product similar results. Bed start temp is 35, First change is to 75, then next is to 0, the next change is back up to 75.

Other users on the discord have tested what i've found and are getting the same results.

npoulos, 2023-02-05:
Something to note is that the artefact only happens when the bed is actively heating, or holding a higher temperature. 

When the bed is turned off/cooling even from a very high temp the artefact completely goes away. 

I don’t think this is being caused by a thermal issue but rather when the bed heater is “on”

pseudo-judo, 2023-02-05:
[image]

can also confirm the issue arises on my printer when the heatbed is active at 65C. The spiral artifact is not visible on 30mm cylinders but is very visible on 20mm cylinders when printing in spiral vase mode.

Pro-100P1P, 2023-02-05:
Same here, terrible leadscrew pattern on small diameter cylinders... 
[image]

npoulos, 2023-02-06:
@SaltWei will this be passed on to the printer team? Is there anywhere I can follow up?

SaltWei, 2023-02-07:
Have been transfered. Will make you updated.

SaltWei, 2023-02-09:
> @SaltWei will this be passed on to the printer team? Is there anywhere I can follow up?
Response from the printer team:
This is the thermal deformation caused by the hot bed heating moment, which has little influence on the printing accuracy

npoulos, 2023-02-09:
@SaltWei this issue is causing massive layer inconsistencies. Is there someone within the printer team I can talk directly to? There is a thread on discord with hundreds of people having layer inconsistencies and it seems this might be causing the issue

pseudo-judo, 2023-02-09:
@SaltWei for the team to say it doesn’t affect accuracy seems odd. We are talking about an effect on print quality which you can clearly see is affected. Yes the print is an accurate 20mm in diameter but you can see it’s covered in spiral artifacts

Wolf-on-Air, 2023-02-09:
There may be multiple issues at work here, and I'm running October firmware for context, that said, this was my payoff from clearing the "on layer change" gcode macro in machine profile entirely, because something in that block trips up the motion planner to come to a full stop several times as it's doing whatever. I am currently unable to test newer firmware versions, unfortunately. My thinking is that the same defect is what's actually happening to cause the precise spiral, if triggered by different means internally. The same defect is caused by M201 and M205 gcode commands to change acceleration and jerk dynamically, and can be removed by slicing at a constant acceleration and jerk for all line types (I used 3000 and 7).

Please disregard the difference in layer perimeter quality as this is comparing 0.05mm layers with 0.2mm layers, but note the absence of spiral artifacting.
[image]
after
[image]
[image]

As it shows up based on temperature setpoint this is unlikely to be the root cause of the error actually described in this ticket, but I wonder if something about the temperature control loop is acting differently and causing the gcode processning to stall. Perhaps runaway checking? 😆