bambulab/BambuStudio #1247: Seam improvement like in the Softfiver fork

Itebayo, 2023-02-05:
Seams are looking pretty bad even with the right pressure advance and flow value, especially on round objects.
In the Softfiver fork there are several options to improve the seam which help quite a lot.
I really think this should get more attention as more and more people are not satisfied with the seam quality and this is mostly a software issue.

The seam alignment on round object is also very poor.
For commercial use bad looking seams are quite the issue.

SaltWei, 2023-02-06:
> In the Softfiver fork there are several options to improve the seam which help quite a lot.

Could you share what options can help? Thank you very much!

Itebayo, 2023-02-06:
Sure!
The option "Seam gap" seems to help quite a bit, if you reduce it form 15% to 1%.
Going all the way to 0% introduces weird travel moves all over the place, which seems to be a bug.

Additionally "wipe on loops" can be turned on and reducing wipe speed to 10% also makes a difference.

All of this was tested with PLA and the Softfiver version 1.4.4 with the bambulab X1.

I have owend quite a few printers and used a lot of slicer software, the seams with the Cura slicer were always the best, so maybe look into what they are doing too. I think the addition of a "coasting feature" would reduce the bulge at the end of a travel move/seam even more.

CCS86, 2023-02-09:
> Sure! The option "Seam gap" seems to help quite a bit, if you reduce it form 15% to 1%. Going all the way to 0% introduces weird travel moves all over the place, which seems to be a bug. [...]

Seam gap is the equivalent of coasting.

Itebayo, 2023-02-09:
From my understanding it is not.
Coasting stops the extrusion near the end of the travel path.
Seam gap leaves a travel gap between the start and endpoint of the seam.

CCS86, 2023-02-09:
> From my understanding it is not. Coasting stops the extrusion near the end of the travel path. Seam gap leaves a travel gap between the start and endpoint of the seam.

In SuperSlicer the extrusion is stopped, but a travel move continues to the original end point (ie coasting). Here is a 300% seam gap:

[image]

Itebayo, 2023-02-09:
Ok looks like my understanding was wrong, is it the same for cura?

CCS86, 2023-02-10:
Yes, Cura stops extrusion but completes the path as well.

npoulos, 2023-04-21:
Coasting in Cura can be controlled and turned off. Seam gap cannot be turned off unless using Orca Slicer.

Given proper PA and LA tuning, the seam gap (and coasting) cause a much bigger gap than would be produced without it.

Please incorporate the seam settings that Orca slicer has. It is the main reason I am not using BambuStudio.

Turning off "Wipe on loops" in orca slicer also greatly improved seams.

All my tests are with a 0.2 nozzle which show great improvement!

npoulos, 2023-04-21:
Also to add - In Cura, you can control coasting amount, speed and minimum extrusion before it coasts. The intent of coasting originally was to control pressure in the nozzle to stop over extrusion at the end of a loop.

The seam gap in BS is a hard-coded 5% gap. It uses the travel speed once it stops extruding and has no rule around when to, and not to add the gap in. It was originally added in Slic3r as a hard-coded coast move but since LA and PA exists it no longer is relevant and causes more issues.

The seam gap also clashes with the wipe move if turned on.

QingZhangBambu, 2023-04-23:
@Itebayo if you don't mind, could provide the stl and 3mf you used? I am testing these features. And has no improvement.

QingZhangBambu, 2023-04-23:
And about "wipe on loop", We've moved that travel path from the 1.4 version.

QingZhangBambu, 2023-04-23:
About "wipe on loop", the path of travel has been moved since version 1.4.

QingZhangBambu, 2023-04-23:
https://github.com/bambulab/BambuStudio/commit/a7266286579a4a81f83f8a3fd47524892c4fe645

QingZhangBambu, 2023-04-24:
There are results of 0.4 nozzle and 0.2 nozzle, sliced by Orca, Printed by Bambu X1C. @Itebayo 
[image]
[image]

arroyoDev, 2023-06-21:
im printing coasters (100mm round objects).   set seam gap to random to hid the ugly line, but still got lots of ugly extrusions where each seam started.   I used Orca (softfiver fork) and reduced the seam gap to 5% with great results.   will work on the perfect % without hitting zero (per comments above).  
[image]

QingZhangBambu, 2023-08-01:
Anyway, we provided those parameters on version 1.7.

closed 2023-08-01 by QingZhangBambu

maurice-lef, 2023-11-16:
> There are results of 0.4 nozzle and 0.2 nozzle, sliced by Orca, Printed by Bambu X1C. @Itebayo [image] [image]

@QingZhangBambu 

how have you done that, that you change the settings at specific points. As far as I know there is no possibility to do so in the OrcaSlicer. I am trying to do something similar but it always takes me a long time as I manually have the rewrite the gcode.

Would be very grateful if you could share your workflow/ strategy.

Thanks in advance. :)

QingZhangBambu, 2023-11-17:
@maurice-lef i change the code, apply it to region parameter, so it can be edit by modifier. it is a method to do quick test. but not suit for you.