bambulab/BambuStudio #1248: AMS rollover support for P1P

Namehater, 2023-02-05:
When using the P1P with AMS, when a roll of filament runs out I want the next slot of the same material to take its place.

currently the AMS knows that the spool has run empty but does nothing but continue printing with no filament.

SaltWei, 2023-02-06:
Not slicer issue.

I have transfered this ticket to printer team.

For issue about firmware and printer, please move to https://forum.bambulab.com/ or BanbuHandy APP to report ticket (https://wiki.bambulab.com/en/x1/troubleshooting/upload-log) for better support. Thanks!

SaltWei, 2023-02-06:
Could you raise a ticket by BambuHandy app(https://wiki.bambulab.com/en/x1/troubleshooting/upload-log)? Because firmware team need log to check this issue.