bambulab/BambuStudio #1250: Incorrect speed for single perimeter multi-part object.

Hologos, 2023-02-06:
**Describe the bug**
I made a multi-part object that consists of cylinders with 1 wall only. Each cylinder is raised by 2mm up and scaled to be 0.4mm narrower than the last one (to create a staircase appearance). If I inspect the speed preview, the very first layer of each part is printed at ~240mm/s, while others at 20mm/s. I tried to tinker with speed settings but could not prevent this behavior.

I inspected the line types and every line type is the same but different speed nonetheless.

**3mf File for This Bug**
[ams-purge-waste-calibration.up-to-8-colors.v2.2.3mf.zip](https://github.com/bambulab/BambuStudio/files/10660327/ams-purge-waste-calibration.up-to-8-colors.v2.2.3mf.zip) is archived with zip.

**To Reproduce**
Steps to reproduce the behavior:
1. Open the project.
2. Slice it.
3. Go to preview tab and select speed Color scheme.
4. See error

**Expected behavior**
The speed would be the same for all layers.

**Screenshots**
[image]
[image]

**Desktop (please complete the following information):**
 - OS: macOS 13.2
 - Version: 1.4.2.13

QingZhangBambu, 2023-02-09:
That's the affection of the cooling strategy. It counts the filament change time into layer time.  So the first change layer's speed would be faster, it hasn't been slowed down. We would evaluate if it needs to change or not.

QingZhangBambu, 2023-02-20:
the time of filament change is counted to layer time, Thus, the layer cooling time will satisfy the layer time. The layer speed will not slow down. That's fine, it will not cause print failure.

closed 2023-02-20 by QingZhangBambu