bambulab/BambuStudio #1252: Anchor points/webbing for tall prints to reduce vibrations and prevent fails

hutje, 2023-02-06:
It would be nice to have a "tree-support"-feature where you can manually place extra "anchor" points to reduce vibration by securing the part with the new way tree-support is generated.

Also with printing multiple parts it would be nice to place Anchor-points between the parts by connecting multiple objects by a few small lines like bridging a "spider web". The lines are easily removeable and do not need to be really cosmetic, but mostly practical.

tangketan, 2023-02-12:
Thank for your idea. The first one is easy, we'll see if it's possible to be included in the next release.
The second one is more complicated. We'll discuss internally.

julie777, 2023-03-17:
I am adding my request to this one as they are both trying to solve the same problem.

Tall prints and especially tall thin prints often fail. Auto brim helps with some prints. There are other things that I do to help with tall prints that might be able to be automatic.

For PLA I turn of the aux fan, so the strong wind doesn't make the model move.
I make the brim 10mm instead of 5mm wide.
I slow down the print speed. (My Prusa prints so much slower that it never fails with tall skinny items.)

What I have noticed is that the rapid motions of the print head always cause some vibration of the printer and the model being printed. I suspect that when moving quickly while extruding there is also some amount of "pull" on the model. The combination of these can cause mis-layering and/or adhesion problems.

I don't have the time to generate all the numbers, but I think that for a huge number of models auto height compensation could prevent print failure. Using some ratio of width/diameter and current height above the build plate speeds could be adjusted.

I know some people print items laying down, such as an axle or shaft, but that does not produce a round part. An 8mm diameter shaft that is 50mm long will be almost perfect standing on end - if it doesn't fail.

fangly, 2023-09-20:
I just had issues printing some tall pieces that kept falling over... A raft with 90% density did not help since it did not adhere well enough to the plate. Increasing the density to 100% helped with plate-raft interface, but then it is the interface between the raft and model that failed. And I did not find a way to make this raft-model stronger in Bambulab. Obviously, I tried setting some manual tree supports, but this seem only possible for overhangs (Bambu Studio 1.7.4.52). My "solution" was adding to the plate some thin disk overlapping with the model to print.

Conclusion: I am totally in favor of adding features to improve model stability, like the one proposed here. But it could also be nice to add options to strengthen rafts.