bambulab/BambuStudio #1253: Enabling support on unrelated region would turn bridges into bottom surfaces.

eddydas, 2023-02-07:
1. Open the attached .3mf model
2. Ensure support is entirely disabled (enable support = false)
3. Slice
4. Observe that the bottom side of the middle "support bar" is sliced as "bridges". This is the expected behavior. See screenshot no_support_bridge.jpg 
[image]

5. Now, enable support (normal, manual)
6. Paint support onto totally unrelated region
7. Slice
8. Observe that the bottom side of the middle "support bar" is now sliced as "bottom surface", with Z shape path, that would definitely fall down due to gravity. See screenshot 2. See end product picture.
9. 
[image]

[image]

macOS. BambuStudio v1.4.2.13

[iPad Stand v3.3mf.zip](https://github.com/bambulab/BambuStudio/files/10670208/iPad.Stand.v3.3mf.zip)

pennystockss, 2023-02-07:
Yes this is very frustrating. Prusa slicer has this bug, at least for soluble supports. I've had to resort to adding an object modifier and basically creating my own bridges where it does this with changing the angle of the infill and such.

eddydas, 2023-02-07:
I've sort of workaround it by changing "bottom surface pattern" to concentric, which would make those bottom surface to become straight lines again... but it's not ideal since it uses a different speed than bridges I think.

pennystockss, 2023-02-07:
Good to know! I figured bottom surface only did first layer pattern. Save me some time.

SaltWei, 2023-02-07:
This is a known bug of tree support, and we will fix it in next release.
Currently, when using tree support, you can set the top contact distance to be 0.01,ranther than 0 to workaround this issue.

SaltWei, 2023-03-01:
Available in v1.5.0 Public Beta

closed 2023-03-01 by SaltWei