bambulab/BambuStudio #1254: Reload Disk loses local coordinates and messed up model

pennystockss, 2023-02-07:
**Describe the bug**
When reloading from disk on a multi part object, all the parts fall apart and lose their local coordinates. Requires reimporting object again.

**3mf File for This Bug**
Hard to attach 3mf file, needs original STL files.

**To Reproduce**
Steps to reproduce the behavior:
1. Load a multi part object (by dragging 2+ parts to the build plate and hit Yes for multi part object)
2. Right click multi part object on the build plate and hit reload from disk.

**Expected behavior**
To reload from disk properly.

**Screenshots**
[image]
[image]
[image]

**Desktop (please complete the following information):**
 - OS: Windows 11
 - Version 1.4.2.13

zhimin-zeng-bambulab, 2023-02-09:
Could you provide the original stl files, it can help us better reproduce the problem, Thanks!

zhimin-zeng-bambulab, 2023-02-21:
I tried to reproduce this problem with combined stl files, but it failed to reproduce. Could you provide the original stl files?

pennystockss, 2023-02-21:
Here try this one for a test, random multipart object:
https://www.thingiverse.com/thing:2485063

All I do is drop all the STL's on the bed.
Hit yes to "Single object with multiple parts"
Then right click and "Reload from disk"

[image]
[image]
[Multi-Color_Cell_Model_2485063.zip](https://github.com/bambulab/BambuStudio/files/10789233/Multi-Color_Cell_Model_2485063.zip)

SaltWei, 2023-03-02:
Fix is included on v1.5.0 Public beta

closed 2023-03-02 by SaltWei