bambulab/BambuStudio #1255: Allow adding K Factor to filament profile instead of AMS slot

pennystockss, 2023-02-07:
**Is your feature request related to a problem? Please describe.**
Sort of. Having the K Factor on the AMS side seems counter intuitive especially if changing multiple manufactures and types. It should be on the filament side.

**Describe the solution you'd like**
An option to change K Factor in the slicer filament profile instead of the AMS section. I would do it with Gcode but I'm unsure if AMS would override it.

**Describe alternatives you've considered**
Changing Gcode for filament.

**Additional context**

SaltWei, 2023-02-07:
1 One machine may need different K value, even with same filament, after a long time using.
2 Third party printing filaments, with same kind and brand but different batches, often vary greatly in K value.

Apulo, 2023-02-22:
I would like to add to this request.

The ability to set K value in the filament profile would really be nice but also keeping the K value override in the AMS section would also be great to override the profile.

RyanEwen, 2023-07-21:
I find it too frustrating to document the K factor for each of my filaments myself and then have to look them up and copy them into the Device tab / AMS area of Bambu Studio every time I change spools.

As a workaround, I have been adding Start G-Code with the setting for each of my filaments:
```
M900 K0.020; adjust linear advance
```

I wish it were simply a setting in the filament profile area as it is in Orca Slicer.

@SaltWei I understand that it may differ per-printer and per batch, however with the P1-series we have to find these values manually and yet there is nowhere to store all these values within Bambu Studio. I like the idea of being able to override it per-printer from the Device tab if-needed, but a core value should exist in the filament profile as a starting point.

softwareByAndi, 2023-07-29:
I agree with storing the k-factors in the filament profiles

# problem
I'm finding my self printing a single layer calibration square sheet in order to fine tune the k-values.

This is because a single line isn't enough data to select the correct value from -- a lot of them look good as a line, but terrible when printing.

# thoughts
another thought is that machine k-factor deltas are likely to be consistent? for example M1 might have higher k-factors

# possible solutions
### 1. consider storing 3 k-factor deltas
1. the base k-factor should be stored in the filament profile`
2. a delta k-factor should be stored in the machine to `fine-tune the filament k-factors for that specific machine`
3. another delta k-factor should be available to edit per object. this should allow us to print multiple objects on a build plate with the same filament, but with slightly different k-factors. _(this is really what I want for building fine tuning calibrations)_

``` javascript
// k = sum of deltas
K = filament_base_k + (object_k_delta || 0) + (machine_k_delta || 0)
```

### 2. filament profiles store a list/dictionary of k-factors (1 for each machine)
``` javascript
// filament profile
{
  ...
  "k-factor": {
    "Machine_001": 0.02,
    "Machine_002": 0.025,
    "Machine_003": 0.015
  }
}
```

### 3. at the very least, please add another k-factor fine calibration (similar to the existing flow rate calibration fine-tuning)
print 7 square sheets with a 0.005 k-factor deltas
- 0=K should be the selected k-factor from the first calibration
- 3 less than: K-0.005, K-0.010, K-0.015
- and 3 greater than: K+0.005, K+0.010, K+0.015

RyanEwen, 2023-07-29:
@softwareByAndi you might like the PA Tower calibration print bettee than the lines. OrcaSlicer had it built in. It's a lot easier to read (especially if the filament color matches the bed) and a lot easier to clean up. Does requires measuring, though.

softwareByAndi, 2023-07-29:
> @softwareByAndi you might like the PA Tower calibration print bettee than the lines. OrcaSlicer had it built in. It's a lot easier to read (especially if the filament color matches the bed) and a lot easier to clean up. Does requires measuring, though.

nah, the PA Tower shows the same data as the lines test. The sheets show 2 important data:
1. size of gaps where infill intersects walls
2. imperfect extrusion mess when changing directions. -- the edges tend to have higher flow per second than the middle, so we either get a very clean center with messy edges, or clean edges with an  under extruded center .

richard-scott, 2023-08-21:
I have just had to reset my printer back to factory settings as it was complaining about something... This now means that I have to re-calibrate all my K values again. It would be great to be able to store these in BBL and push them to the printer.

EklDuderino, 2023-08-27:
Throwing support in for storing K factor values in filament profile setting. Even if just a note section to reference.

GuzziRaz, 2023-09-12:
Whatever solution you come up with here also need to address the issue that all K values are forgotten at poweroff.

I mean, you can even resume a print that halted at a power loss... BUT you'll resume it with all K changed to 0.00 mid-print?! Come on.

DerrillWMS, 2023-10-02:
Just to confirm, K values are **NOT** overridden in the X1C's firmware when a print job w/ filament profile(s) is sent to the X1C, correct?

threadstone83, 2023-11-10:
I would also prefer when the k values are saved in the filament profiles. It's annoying to write down the values by myself somewhere and enter them after every change.

Distracto069, 2023-11-25:
I back up this idea. Come on bambu release an update

GuzziRaz, 2023-11-26:
> Whatever solution you come up with here also need to address the issue that all K values are forgotten at poweroff.
> 
> I mean, you can even resume a print that halted at a power loss... BUT you'll resume it with all K changed to 0.00 mid-print?! Come on.

I have found out that if I store a PA calibration with its **default name** (eg. `Bambu PLA Basic`) it will survive a power cycle, but if I (as actually suggested by calibration) add color to the name (eg. `Bambu PLA Basic orange`), it will not (the profile is there but I have to manually enable it). That's a sad shortcoming as different colors tend to get different values. My current workaround is to store the most common result (such as 0.020) with the default name and all other results with an amended name. Some documentation of this current behavior would have saved me some times and some suboptimal prints 😢 and a fix so the printer remember the non-default settings would be welcome.

MrDB42, 2024-01-18:
Here's a thought with working with Bambu brand filament with RFID tags - associate K values and flow rate values to the RFID number on the filament roll. Since the printer has an odometer, once the roll has been used it automatically deletes that information from the printer. Or if you don't use it up and have left over filament you know you won't use, be able to delete that information manually.

Even better would be able to write the K values and flow rate to the RFID tag. That way you can takeout then later put it back in later or move rolls of filament in the AMS and not have to worry remembering what value goes with what spool. But I believe that would require an updated AMS unit so the RFID reader board can push the information to the writable RFID tags on the spool. But that could be a newer upgraded AMS unit and updated RFID tags on the spools.

P.S.
I would be nice if Bambu would allow other filament manufacturers to use RFID tags on their filament also (with a licensing fee if needed).

DanielLaberge, 2024-02-26:
+1 for actually saving the K-Value in the filaments profile. The current behaviour is user-hostile and not befitting the level of automation Bambu Labs printers are capable of.

Thank you for the consideration.

bcarriveau, 2024-03-30:
Please make this happen!!

Hyrules, 2024-06-03:
I add my voice to adding the K-factor to the filament profile.  it is really annoying to have to reconfigure it every time I restart my printer (X1C). I tend to turn off my printer when not using it and I often forget to reconfigure the k factor when starting it back. Adding it to the profile would be a set it and forget it solution. Maybe hidden in the advanced settings or something similar.

richard-scott, 2024-07-22:
Isn't this already in the filament profile? But labelled Pressure Advance?
Or is BBL's k Value something different?