bambulab/BambuStudio #1256: Add AMS option to continue print with a different spool of filament

guillaumemolter, 2023-02-07:
**Is your feature request related to a problem? Please describe.**
When filament runs out in one slot, and you don't have a second exact spool of filament, the only way to keep printing is by either loading the current empty slot with a new spool or move another spool from another slot to the current one.

**Describe the solution you'd like**
I would like an option to select a different AMS slot to continue the print. ie: I started printing with blue PLA in slot 1, blue PLA runs out, I can select from the screen a different slot to conitnue printing for example red PLA in slot 2.

**Describe alternatives you've considered**
Have a second checkbox in AMS settings below "AMS filament backup" that says "ignore color and brand". In other word if I'm printing in PLA continue with any other spool of PLA loaded.

**Additional context**
Add any other context or screenshots about the feature request here.

morgano86, 2023-02-07:
Had this issue today, printing in Bambu PLA and not "Generic PLA" therefore no auto switch. Had to remove the empty spool and move the "Generic PLA" from a different slot. Pointless action when the AMS could have loaded it for me directly from the slot it was in.

iandouglas, 2023-02-08:
My workaround on this is to set the brand/color on the printer before a print starts to say they're all the same brand/color then auto-runout switch works great. I agree, it would be nice to say "any other slot with the same material type (PLA, PETG) would be great, but I can also understand from their viewpoint that not all PLA settings are necessarily the same from one manufacturer to the next.

guillaumemolter, 2023-02-11:
@iandouglas thanks! I can't believe I didn't think about this workaround... if the argument is truly different filament have different settings then they need to give us either a lot more filament profiles or custom profiles. Anyway thanks for tip!

tsdexter, 2023-02-18:
How does the auto run out work? I had two "generic PLA" loaded in slot one and two, both eSun PLA+, same color and when slot one ran it out it didn't let me choose slot 2 or select it on its own, the only option was to load a new roll into slot one.

This just happened yesterday and I just updated firmware the day before. 

My proposed solution would be on the "send to printer popup" where you select which slots for which colors there is a second line for backups and you can select which slot to use as a backup for each color as well.

Merlwynd, 2023-02-18:
Works for me when I select auto switch on printer options screen. Then have to make sure that on AMS screen the filaments are exactly the same selection, including color, and the same for the filaments selected in Bambu Studio. You can of course physically put any color of filament in the second slot, as the AMS won't know (Unless using Bambu spools that have RFID tag)......

Fireflaker, 2023-04-09:
Please at least allow manual remote unload reload of a different filament slot when printed is paused for example

StoneLiBambulab, 2023-04-11:
hi @guillaumemolter ,
We plan to display and edit the backup relationships between filaments in ams slot. It will take some time to modify with the printer firmware. The release version will be about 2 months later.

Thanks!
Stone Li

arhi, 2023-04-15:
I'm just having the same issue :( ... green bambu pla run out and there's no way I can make it continue with blue bambu pla in the second slot and I'm not going to be near that printer for 3 days and it's a 9h print stopped at 50% :( ... I assumed it will auto-continue using another PLA ...

hella909, 2023-04-30:
I had same problem when PLA run out in the middle of the printing. I expected the AMS to pick up same brand/color filament from the next slot, but that never happened. When I tried to replace the empty spool with a new one, the AMS froze up and wouldn't let me even pick up color or slot. Ended up with ruined print. Luckily enough it was just a box with a lid (lid already was done by that time) and I was able to save the project just ironing unfinished top layer. Looks good, but 50% shorter than I planned originally. Love my X1 Carbon though!!!

hella909, 2023-04-30:
> hi @guillaumemolter , We plan to display and edit the backup relationships between filaments in ams slot. It will take some time to modify with the printer firmware. The release version will be about 2 months later.
> 
> Thanks! Stone Li

Hopefully you are working on this option and we'll see it in the nearest future. So much potential in theses machines, still can't wrap my mind around the fact how great it is. I'm coming from my previous experience with Creality Cr-10 and Prusa MK3...

bookofthefuture, 2023-05-07:
> hi @guillaumemolter , We plan to display and edit the backup relationships between filaments in ams slot. It will take some time to modify with the printer firmware. The release version will be about 2 months later.
> 
> Thanks! Stone Li

Really needs this feature. It's kinda baffling you can't get it to resume from same filament in another AMS slot already. And I can't find any auto-runout options in the Print Options?

tomandegg, 2023-07-21:
@StoneLiBambulab is there any update on this feature? I would like to have the option of continuing printing with a different color of the same material, too. For lots of parts I don't care about the color. And also, if I have only a little of one color left, I'd like to use up all of it instead of splitting up a print manually to minimise waste.

Merlwynd, 2023-07-21:
Just in case folks don’t know, but a simple workaround is to “lie” to the AMS but specifying all your filament spools to be the same Color, etc….then loading the spools you actually want (won’t work with Bambu filament with RFID tag…)

walterwongbbl, 2023-08-02:
> 此功能是否有任何更新？我也希望可以选择继续使用相同材料的不同颜色进行打印。对于很多零件，我不在乎颜色。而且，如果我只剩下一点点颜色，我想用完所有颜色，而不是手动拆分打印件以尽量减少浪费。

Please use the latest version of Bambu Studio

walterwongbbl, 2023-08-02:
> @StoneLiBambulab is there any update on this feature? I would like to have the option of continuing printing with a different color of the same material, too. For lots of parts I don't care about [...]

Please use the latest version of Bambu Studio

walterwongbbl, 2023-08-02:
Please use the latest version of Bambu Studio.
We have already supported alternative use of the same filament.

closed 2023-08-02 by walterwongbbl

tomandegg, 2023-08-05:
@walterwongbbl Could you please elaborate how auto refill with the same kind of filament but different color is supposed to work? I am running the latest version of Bambu Studio but as far as I can see auto refill only works with the exact same filament, i.e., including color.

guillaumemolter, 2023-08-07:
@StoneLiBambulab @walterwongbbl this needs to be re-opened. Just like @tomandegg I'm running the latest version of Studio and latest firmware and this is still not an option...worst the message telling me it's not an option is not even readable anymore (see how it cuts on the right side).
[image]
:

Buzzology, 2023-08-21:
Same, doesn't seem to be working for me either.

raleighlittles, 2023-10-15:
> My workaround on this is to set the brand/color on the printer before a print starts to say they're all the same brand/color then auto-runout switch works great. I agree, it would be nice to say [...]

How do you do this?

> set the brand/color on the printer before a print starts to say they're all the same brand/color

MachineMonkey, 2023-10-22:
set the brand/color on the printer before a print starts to say they're all the same brand/color

But I'm pretty certain I'm doing that and it simply isn't working.  Any chance you could show us what you mean?

Philosophist, 2023-11-17:
My issue is highly related. The spool of Atomic filament is just barely too big for the AMS and the PLA is worn down by the motors trying to fix the problem automatically, the result is that even after I hold the spool and feed it by hand, it won't go through, I have to cut it. There is another factor I'm just learning about where this spool just doesn't get spun effectively even when I open the lid of the AMS and it inevitably jams as I leave it. I've tried loosening it regularly, only to find it tangling and wearing down the filament AGAIN. 

Rather than waste 2ft of filament again by cutting it, I tried to move the filament to the outer spool holder to finish the print and the device refuses to acknowledge the change midprint. Why can't we edit filament settings midprint, exactly? This seems unnecessarily obtuse and easy to fix. I'm about to waste another large length of filament. PLEASE FIX. That said, I am REALLY happy with how far Bambu has gone so far to make so much of these problems automatically fix themselves, so, please raise the priority of this issue and keep up the good work!

dsloyer74, 2024-01-12:
I'm having a similar issue with AMS Lite on the Bambu Labs A1.  I ran out of WHITE Bambu PETG basic during a long print and want to resume with BLACK Bambu PETG basic.  This is the same brand with same flow characteristics.  I prime / paint most of our prints for production and color is not often a concern beyond post-processing options (black is better for metal/luminescent paints).  I should have the option, especially with Bambu filament, to switch color.  

This seems like a gimmick to sell more filament that is not required.  Bambu filament is great and I would purchase anyway, but I should not be required to stock multiple rolls of each color to continue a critical print job.  I already own many print rolls that I need to work through and we need to minimize waste to keep production costs low.

eried, 2024-01-17:
Yes, please, add "continue anyways" to another spool, even if is a manual selection when the same spool is not found

Grazfather, 2024-02-08:
This issue / request hasn't been fixed. What we want is the ability to continue a print with another spool. We cannot change was is loaded, and it won't let us select another spool, even if it's identical in everything except for the colour.

RickP, 2024-03-02:
Why is this closed? I have to manually take the spool from another slot and put it in the slot of the empty spool. That is stupid.

eried, 2024-03-03:
> Please use the latest version of Bambu Studio. We have already supported alternative use of the same filament.

Not true, please reopen this issue :) as @RickP says it is very stupid to have to manually switch spools position just to resume

guillaumemolter, 2024-03-03:
@walterwongbbl @StoneLiBambulab Please re-open this issue, this is not solved.

seanbetts, 2024-03-10:
I could really do with this feature too - @walterwongbbl @StoneLiBambulab please re-open!

tsdexter, 2024-03-11:
I created a new issue as I doubt they pay much attention to new comments on closed issues... comment/upvote there to get it through https://github.com/bambulab/BambuStudio/issues/3703

@seanbetts @guillaumemolter @eried @RickP @Grazfather

kohoutbrady, 2024-04-01:
Assuming no traction on this? New to Bambu (about 1 week with a P1S) but this feature seems like a no brainer to me. I have tons of half empty spools I would love to have the option to manually pick which spool is next.

nguye1p, 2024-04-29:
Also waiting on this feature, manually switching spools is so infuriating.

guillaumemolter, 2024-04-29:
@kohoutbrady @nguye1p please post on the new issue:  this https://github.com/bambulab/BambuStudio/issues/3703

bartcerneels, 2024-10-11:
Adding my +1 to this feature request.

Ran into this today: a matte grey spool ran out on the last 5% of a functional print and I have matte black in another AMS slot. I'm out for the night so would like to continue the print remotely using the matte black but there is no option to force it.
Please give us this choice Bambu. It's a huge benefit of the AMS and your ecosystem with the handy app and connected firmware.

djp55, 2024-10-21:
I figured out how to do this in Bambu Studio - as far as I've seen you can't do it on the fly, but you can set it up to fail over to a different spool of a different color.  In Prepare, go to objects, click your plate, go to plate settings, Other layers filament sequence, and then you can customize your sequence of spools.  

[image]

[image]

sperman32, 2024-10-31:
> Adding my +1 to this feature request.
>
> Ran into this today: a matte grey spool ran out on the last 5% of a functional print and I have matte black in another AMS slot. I'm out for the night so would [...]

Another +1.

lhaley2011, 2024-12-28:
Another +1. Reopen please!

cpomerantz, 2025-01-24:
Why is this difficult to implement? Seems like no different than an automatic change in filament. Just let us switch the spool being used and/or unload and load a spool.

tsdexter, 2025-04-16:
@sperman32 @lhaley2011  @cpomerantz  @steve-a-jones comment/upvote on #3703 which is still open

odama626, 2025-10-26:
#3703 is not the same. I want to be able to say if you run out of #4 use #2

atomarch, 2026-11-01:
Another +1. Would really like to see this feature added, thanks!