bambulab/BambuStudio #1257: incorrectly generated supports - Tree(auto) -> Tree Slim

MarcoPiazza83, 2023-02-07:
incorrectly generated supports 
Type: tree (auto)
Style: tree slim
"only on construction plate" activated

 - OS: Mac os Ventura 13.1
 - Version bambu studio 1.4.2.13

https://user-images.githubusercontent.com/4783556/217346954-2b96f580-8209-4893-8ac6-59bd77f1a090.mov

[image]

[image]

[image]

Kwintyy, 2023-02-07:
Similar issue here,

[image]

[Untitled_plate_1.gcode.3mf.zip](https://github.com/bambulab/BambuStudio/files/10679246/Untitled_plate_1.gcode.3mf.zip)

tangketan, 2023-02-09:
Would you please share the 3mf? gcode.3mf doesn't contain the model so it's not suitable for diagnosis.

tangketan, 2023-02-16:
I confirmed that the "build plat only" option was broken in the current public version, which will be fixed in the next release.

SaltWei, 2023-03-01:
Available in v1.5.0 Public Beta

closed 2023-03-01 by SaltWei