bambulab/BambuStudio #1261: Printer apparently corrupting SD Card partition when print is cancelled

tomservaux, 2023-02-08:
I use this printer for my business and have had it for less than 3 weeks. When I try to print after cancelling a previous print from Bambu Studio through the cloud, at least since updating to the latest firmware, the printer will almost always stop at a random point mid-print. The light turns off but all of the fans continue. The UI on both the printer and Bambu Studio show that the printer is printing, not paused or cancelled. From what I can tell it never recovers. I have switched SD cards 3 times with brand new cards formatted on the printer as recommended. If I reformat the card on the printer after this happens it still fails on every print afterwards. Once I move the SD card to my PC, recreate the volume on the card and then have the printer format it the printer appears to be able to complete prints until I cancel one again. 

So far this has been reproducible almost 100% of the time for larger prints, to the point where I have started recreating the volume on the SD card every time I have to cancel a print. I have seen it finish short prints, like a test cube, and it does not appear to happen at the same point for all prints, but it does seem to happen at the same point if you restart the same print.

Steps to reproduce the behavior:
1. Run a print from Bambu Studio
2. Wait until the print finishes downloading to the printer
3. Cancel the print from Bambu Studio.
4. Start the print again from Bambu Studio
5. The print will start but will stop at some arbitrary point. 

**Expected behavior**
The print should finish successfully every time unless there is user error.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Desktop (please complete the following information):**
 - OS: Windows 11
 - Bambu Studio 1.4.2.13 
 - Printer and Firmware Info: 
![image](https://user-images.githubusercontent.com/1427978/21760647
[image]
[image]
6-1438b13f-6c63-4b9d-842e-452a15547937.png)

tomservaux, 2023-02-08:
Latest logs in logger directory on the SD card
[01S00A2B0702143_02-03_15_09_07.788_idx_26.log](https://github.com/bambulab/BambuStudio/files/10688476/01S00A2B0702143_02-03_15_09_07.788_idx_26.log)
[01S00A2B0702143_02-08_22_51_55.357_idx_24.log](https://github.com/bambulab/BambuStudio/files/10688478/01S00

Latest core log file
[corelogger-vr00.02.10.39_01S00A2B0702143_20230203150907_e.zip]
(https://github.com/bambulab/BambuStudio/files/10688483/corelogger-vr00.02.10.39_01S00A2B0702143_20230203150907_e.zip)

The gcode of the file I was printing
[bblgcode.zip](https://github.com/bambulab/BambuStudio/files/10688502/bblgcode.zip)

The 3mf file I was printing. 
[bbl.zip](https://github.com/bambulab/BambuStudio/files/10688498/bbl.zip)

A2B0702143_02-08_22_51_55.357_idx_24.log)

SaltWei, 2023-02-09:
This is impossible to be bug of slicer. Mark as printer issue and sent to printer team.

SaltWei, 2023-02-09:
By the way, for future issue about firmware and printer, please move to https://forum.bambulab.com/ or BanbuHandy APP to report ticket (https://wiki.bambulab.com/en/x1/troubleshooting/upload-log) for better and faster support.
Thanks!

DanTup, 2024-05-13:
I had this issue after a cancelled print too. I was unable to revive the SD card with Windows or MacOS. Everything reported that it was read-only (even if I tried using diskpart to clear any readonly flags). Had to replace the SD card (plus I got some spares in case it happens again!).

RheaAyase, 2024-12-19:
+1, brand new A1 corrupted the included SD card an a completely different subsequent print threw an error which was visible only on the display and all my apps were thinking that everything is fine with no errors. Yet another 12h print ruined for no good reason :(