bambulab/BambuStudio #1262: Allow users to customize the "go live" ffmpeg output filename/location

iandouglas, 2023-02-08:
**Is your feature request related to a problem? Please describe.**
I love the "Go Live" feature. But it's limited to a single printer, and I'd like to make a way to monitor several printer cameras on my network. Currently, Bambu Studio only allows one printer to go live at a time.

**Describe the solution you'd like**
I'd love an option in the "Go Live" setup to allow the user to choose the output filename (and location) of where Bambu Studio would create the ffmpeg.sdp output stream file so that multiple printers could be streamed at the same time if the computer can handle the task. (if not, that's on the user, not BS or ffmpeg)

Ideally, instead of having to go through each printer manually to "go live" on each one, I'd love a panel that shows all of my printers, with a "go live" toggle next to each one, with a customized filename/location for the SDP output. The output could be a local HTML page that loads in a browser with an embedded video player for each of the streams.

And since we're doing this on a per-printer basis as well, perhaps something similar to the "Global/Object" toggle when printing several parts, maybe have global settings for things like resolution, format, etc, that could also be overridden on a per-printer basis. Maybe one printer is working on a super-detailed print where I want a 1080p output, and another is in draft mode where I only care that it's not spaghetti so 720p or lower resolution is just fine.

**Describe alternatives you've considered**
I've wondered whether renaming the stream filename while streaming would maintain the file pointer or not (I'm guessing it'd throw an error), but that's still a manual effort and if you have a fleet of printers this could be a pretty laborious task.

**Additional context**
I love that the team chose the open-source ffmpeg utility for this. I have a pretty beefy PC (32-cores, 128GB of RAM) so I can handle streaming several printers if Bambu Studio would allow me to do that.

bambu123, 2023-03-02:
Thank you for the suggestion. We will consider it.

ichihollow, 2024-03-07:
YES!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!  Please don't make the community rip your software apart to create the feature, then we end up having to risk our systems to try and get the functions we crave.  I have two of the X1Cs and I know that unless you release new version of the X1C I'm probably going to end up with at least 2 more by the end of the year.  I'm DESPERATE for this functionality!  My only other option is to share network folders on multiple computers to get the same result.

cmguo, 2024-03-08:
I think this may help.
#2915