bambulab/BambuStudio #1265: STL support export negative parts

zhimin-zeng-bambulab, 2023-02-09:
issue from #1216

theogott, 2023-02-09:
Please see also:
[https://github.com/bambulab/BambuStudio/issues/1156][11](https://github.com/bambulab/BambuStudio/issues/1156))

tensoralex, 2023-11-30:
Same for me, issue.

marcodent, 2024-04-08:
same issue here. in 2024

zhimin-zeng-bambulab, 2024-04-08:
[image]

You can execute meshboolean and then export to stl.

marcodent, 2024-04-08:
If I apply Mesh boolean, I get an error and the negative part is lost.
[image]

SaltWei, 2024-06-05:
[image]
Already supported when export stl with negative part.

Shadewarp, 2024-06-13:
Not completed - it still doesn't work - it won't allow me to perform mesh boolean before exporting, due to only positive parts being kept and a part is not bound a volume (whatever that means)

[image]

Phynixx, 2024-10-15:
Same for me

invisibleninja06, 2024-11-05:
> Not completed - it still doesn't work - it won't allow me to perform mesh boolean before exporting, due to only positive parts being kept and a part is not bound a volume (whatever that means)
> 
> ![image_2024-06-13_113250334](https://private-user-images.githubusercontent.com/6478994/339304994-9941cf41-01ce-43e6-bb06-2fad4aa80e25.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MzA4MjAwMjMsIm5iZiI6MTczMDgxOTcyMywicGF0aCI6Ii82NDc4OTk0LzMzOTMwNDk5NC05OTQxY2Y0MS0wMWNlLTQzZTYtYmIwNi0yZmFkNGFhODBlMjUucG5nP1gtQW16LUFsZ29yaXRobT1BV1M0LUhNQUMtU0hBMjU2JlgtQW16LUNyZWRlbnRpYWw9QUtJQVZDT0RZTFNBNTNQUUs0WkElMkYyMDI0MTEwNSUyRnVzLWVhc3QtMSUyRnMzJTJGYXdzNF9yZXF1ZXN0JlgtQW16LURhdGU9MjAyNDExMDVUMTUxNTIzWiZYLUFtei1FeHBpcmVzPTMwMCZYLUFtei1TaWduYXR1cmU9M2JkZWEwODY0MDg0MzE2NGI3YmFlYzYyYzI4N2Q3NWIxZmRjZTYyMjM0ODc2ZDk3Y2NiY2RjMTBkYTZiMDc1NCZYLUFtei1TaWduZWRIZWFkZXJzPWhvc3QifQ.67wYMDescEsv07L-orcZyaJGc5cnInaBOo3co-AHhUM)

Part not bound to a volume in my experience means an issue with one of your objects. When i got that recently i loaded up the stl into MeshMixer and repaired the file, after which i no longer got that error. I would suggest trying that, ultimately would be nice if the Bambu repair could fix volume not bound issues which in my case it didnt.