bambulab/BambuStudio #1268: Filament change: Pull filament back right after the cutting it

rovili, 2023-02-09:
**Is your feature request related to a problem? Please describe.**
Not a problem, but an improvement.

**Describe the solution you'd like**
Currently the pullback of the filament starts when the printhead is back above the "poop-box".

I think it would make sense and also would save time if the pullback starts right after the filament got cut. It would already start to pullback the filament while the printhead is moving to the "poop-box".

rovili, 2023-02-11:
Another thing what would save time:
Remove the nozzle wipe before cutting the filament. Move direct from print object to cut the filament.

lanewei120, 2023-03-02:
thanks for the suggestion

After cutting off the filament, As the nozzle will leak, it needs to be moved to the trash immediately, and then pullback the filament

EukaDesigns, 2023-03-09:
@rovili 
Before retracting the filament, the extruder will actually push out a bit of filament first. So, in addition to oozing as mentioned, the extruder would be pushing filament out while moving if this process happened immediately after cutting the filament. 

As far as the nozzle wipe goes, I believe there is only one unload filament function that is used every time, regardless of when the function is called. This means that there are times it may be good to wipe (e.g. the print was paused and filament has been oozing a bit while waiting for user input - if there was no wipe there would be potential to drag the oozed filament over the print on the way to cutting the filament), so it means it'll wipe every time. If the code is run differently based on circumstances then the change could be made to sometimes wipe and sometimes not, but as it is organized now, it is just safer to wipe every time.