bambulab/BambuStudio #1270: Bambu Studio Crash on Start - Mac Mini M1

bravojohnny1, 2023-02-10:
**Describe the bug**
Bambu Stuio crashes on start. No changes when reinstalling. 

**To Reproduce**
Steps to reproduce the behavior:

OS Ventura on Mac M1
Bambu Studio 1.4.2.13

1. Open Bambu Studio
2. Program crashes

**Expected behavior**
Program should open

Error Report:

-------------------------------------
Translated Report (Full Report Below)
-------------------------------------

Process:               BambuStudio [24935]
Path:                  /Applications/BambuStudio.app/Contents/MacOS/BambuStudio
Identifier:            com.bambulab.bambu-studio
Version:               01.04.02.13 ()
Code Type:             ARM-64 (Native)
Parent Process:        launchd [1]
User ID:               501

Date/Time:             2023-02-10 09:50:39.7202 +0100
OS Version:            macOS 13.0 (22A380)
Report Version:        12
Anonymous UUID:        AFE886FA-7143-4B42-4AA4-A22028C9A226

Time Awake Since Boot: 4600 seconds

System Integrity Protection: enabled

Crashed Thread:        0  bambustu_main  Dispatch queue: com.apple.main-thread

Exception Type:        EXC_BAD_ACCESS (SIGSEGV)
Exception Codes:       KERN_INVALID_ADDRESS at 0x0000000000000000
Exception Codes:       0x0000000000000001, 0x0000000000000000

Termination Reason:    Namespace SIGNAL, Code 11 Segmentation fault: 11
Terminating Process:   exc handler [24935]

VM Region Info: 0 is not in any region.  Bytes before following region: 4340973568
      REGION TYPE                    START - END         [ VSIZE] PRT/MAX SHRMOD  REGION DETAIL
      UNUSED SPACE AT START
--->  
      __TEXT                      102be0000-105ed0000    [ 50.9M] r-x/r-x SM=COW  ...S/BambuStudio

Thread 0 Crashed:: bambustu_main Dispatch queue: com.apple.main-thread
0   ???                           	               0x0 ???
1   BambuStudio                   	       0x103557f3c Slic3r::GUI::GLModel::init_from(indexed_triangle_set const&) + 152
2   BambuStudio                   	       0x1034ff418 Slic3r::GUI::GLGizmoBase::GLGizmoBase(Slic3r::GUI::GLCanvas3D&, std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> > const&, unsigned int) + 356
3   BambuStudio                   	       0x1035005fc Slic3r::GUI::GLGizmoMove3D::GLGizmoMove3D(Slic3r::GUI::GLCanvas3D&, std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> > const&, unsigned int, Slic3r::GUI::GizmoObjectManipulation*) + 36
4   BambuStudio                   	       0x1034f262c Slic3r::GUI::GLGizmosManager::init() + 540
5   BambuStudio                   	       0x103498ad0 Slic3r::GUI::GLCanvas3D::init() + 1268
6   BambuStudio                   	       0x1035d6f2c Slic3r::GUI::GUI_App::post_init() + 2260
7   BambuStudio                   	       0x10360c89c wxEventFunctorFunctor<wxEventTypeTag<wxIdleEvent>, Slic3r::GUI::GUI_App::on_init_inner()::$_26>::operator()(wxEvtHandler*, wxEvent&) + 556
8   BambuStudio                   	       0x1053d8eb8 wxEvtHandler::SearchDynamicEventTable(wxEvent&) + 252
9   BambuStudio                   	       0x1053d8b84 wxEvtHandler::ProcessEventLocally(wxEvent&) + 88
10  BambuStudio                   	       0x1053d89ac wxEvtHandler::ProcessEvent(wxEvent&) + 56
11  BambuStudio                   	       0x10533c1cc wxAppConsoleBase::ProcessIdle() + 92
12  BambuStudio                   	       0x10512ce70 wxAppBase::ProcessIdle() + 24
13  BambuStudio                   	       0x1052caaa8 wxApp::ProcessIdle() + 36
14  BambuStudio                   	       0x1053e1c24 wxCFEventLoop::CommonModeObserverCallBack(__CFRunLoopObserver*, int) + 88
15  CoreFoundation                	       0x1a69d4de4 __CFRUNLOOP_IS_CALLING_OUT_TO_AN_OBSERVER_CALLBACK_FUNCTION__ + 36
[... 551 lines cut ...]
Display: DELL S2721DGF, 2560 x 1440 (QHD/WQHD - Wide Quad High Definition), Main, MirrorOff, Online
Display: DELL S2721DGF, 2560 x 1440 (QHD/WQHD - Wide Quad High Definition), MirrorOff, Online
Memory Module: LPDDR4, Hynix
AirPort: spairport_wireless_card_type_wifi (0x14E4, 0x4378), wl0: Sep  3 2022 03:37:22 version 18.20.379.4.7.8.143 FWID 01-e3c71b50
Bluetooth: Version (null), 0 services, 0 devices, 0 incoming serial ports
Network Service: Ethernet, Ethernet, en0
Network Service: Wi-Fi, AirPort, en1
USB Device: USB31Bus
USB Device: USB31Bus
USB Device: Plantronics Blackwire 5210 Series
USB Device: USB30Bus
USB Device: USB5734
USB Device: hub_device
USB Device: USB Laser Mouse
USB Device: USB2734
USB Device: hub_device
USB Device: MXT USB Device
USB Device: EPOS IMPACT D1
USB Device: Hub Controller
Thunderbolt Bus: Mac mini, Apple Inc.
Thunderbolt Bus: Mac mini, Apple Inc.

**Desktop (please complete the following information):**
 - OS: MAC OS Ventura
 - M1 Chip

lanewei120, 2023-02-11:
> **Describe the bug** Bambu Stuio crashes on start. No changes when reinstalling.
> 
> **To Reproduce** Steps to reproduce the behavior:
> 
> OS Ventura on Mac M1 Bambu Studio 1.4.2.13
> 
> 1. Open Bambu Studio
> 2. Program crashes
> 
> **Expected behavior** Program should open
> 
> Error Report:
> 
> ## Translated Report (Full Report Below)
> Process: BambuStudio [24935] Path: /Applications/BambuStudio.app/Contents/MacOS/BambuStudio Identifier: com.bambulab.bambu-studio Version: 01.04.02.13 () Code Type: ARM-64 (Native) Parent Process: launchd [1] User ID: 501
> 
> Date/Time: 2023-02-10 09:50:39.7202 +0100 OS Version: macOS 13.0 (22A380) Report Version: 12 Anonymous UUID: AFE886FA-7143-4B42-4AA4-A22028C9A226
> 
> Time Awake Since Boot: 4600 seconds
> 
> System Integrity Protection: enabled
> 
> Crashed Thread: 0 bambustu_main Dispatch queue: com.apple.main-thread
> 
> Exception Type: EXC_BAD_ACCESS (SIGSEGV) Exception Codes: KERN_INVALID_ADDRESS at 0x0000000000000000 Exception Codes: 0x0000000000000001, 0x0000000000000000
> 
> Termination Reason: Namespace SIGNAL, Code 11 Segmentation fault: 11 Terminating Process: exc handler [24935]
> 
> VM Region Info: 0 is not in any region. Bytes before following region: 4340973568 REGION TYPE START - END [ VSIZE] PRT/MAX SHRMOD REGION DETAIL UNUSED SPACE AT START ---> __TEXT 102be0000-105ed0000 [ 50.9M] r-x/r-x SM=COW ...S/BambuStudio
> 
> Thread 0 Crashed:: bambustu_main Dispatch queue: com.apple.main-thread 0 ??? 0x0 ??? 1 BambuStudio 0x103557f3c Slic3r::GUI::GLModel::init_from(indexed_triangle_set const&) + 152 2 BambuStudio 0x1034ff418 Slic3r::GUI::GLGizmoBase::GLGizmoBase(Slic3r::GUI::GLCanvas3D&, std::__1::basic_string<char, std::__1::char_traits, std::__1::allocator > const&, unsigned int) + 356 3 BambuStudio 0x1035005fc Slic3r::GUI::GLGizmoMove3D::GLGizmoMove3D(Slic3r::GUI::GLCanvas3D&, std::__1::basic_string<char, std::__1::char_traits, std::__1::allocator > const&, unsigned int, Slic3r::GUI::GizmoObjectManipulation*) + 36 4 BambuStudio 0x1034f262c Slic3r::GUI::GLGizmosManager::init() + 540 5 BambuStudio 0x103498ad0 Slic3r::GUI::GLCanvas3D::init() + 1268 6 BambuStudio 0x1035d6f2c Slic3r::GUI::GUI_App::post_init() + 2260 7 BambuStudio 0x10360c89c wxEventFunctorFunctor<wxEventTypeTag, Slic3r::GUI::GUI_App::on_init_inner()::$_26>::operator()(wxEvtHandler*, wxEvent&) + 556 8 BambuStudio 0x1053d8eb8 wxEvtHandler::SearchDynamicEventTable(wxEvent&) + 252 9 BambuStudio 0x1053d8b84 wxEvtHandler::ProcessEventLocally(wxEvent&) + 88 10 BambuStudio 0x1053d89ac wxEvtHandler::ProcessEvent(wxEvent&) + 56 11 BambuStudio 0x10533c1cc wxAppConsoleBase::ProcessIdle() + 92 12 BambuStudio 0x10512ce70 wxAppBase::ProcessIdle() + 24 13 BambuStudio 0x1052caaa8 wxApp::ProcessIdle() + 36 14 BambuStudio 0x1053e1c24 wxCFEventLoop::CommonModeObserverCallBack(__CFRunLoopObserver*, int) + 88 15 CoreFoundation 0x1a69d4de4 **CFRUNLOOP_IS_CALLING_OUT_TO_AN_OBSERVER_CALLBACK_FUNCTION** + 36 16 CoreFoundation 0x1a69d4cd0 __CFRunLoopDoObservers + 532 17 CoreFoundation 0x1a69d4410 __CFRunLoopRun + 1048 18 CoreFoundation 0x1a69d38a4 CFRunLoopRunSpecific + 612 19 HIToolbox 0x1b00473bc RunCurrentEventLoopInMode + 292 20 HIToolbox 0x1b0047200 ReceiveNextEventCommon + 672 21 HIToolbox 0x1b0046f48 _BlockUntilNextEventMatchingListInModeWithFilter + 72 22 AppKit 0x1a9c2c630 _DPSNextEvent + 632 23 AppKit 0x1a9c2b7c0 -[NSApplication(NSEvent) _nextEventMatchingEventMask:untilDate:inMode:dequeue:] + 728 24 AppKit 0x1a9c1fbf0 -[NSApplication run] + 464 25 BambuStudio 0x105304ffc wxGUIEventLoop::OSXDoRun() + 160 26 BambuStudio 0x1053e22ec wxCFEventLoop::DoRun() + 40 27 BambuStudio 0x10536111c wxEventLoopBase::Run() + 132 28 BambuStudio 0x10533bfc4 wxAppConsoleBase::MainLoop() + 120 29 BambuStudio 0x1052cab04 wxApp::OnRun() + 36 30 BambuStudio 0x10538ef7c wxEntry(int&, wchar_t**) + 148 31 BambuStudio 0x1035d0284 Slic3r::GUI::GUI_Run(Slic3r::GUI::GUI_InitParams&) + 172 32 BambuStudio 0x102bf02c4 Slic3r::CLI::run(int, char**) + 18484 33 BambuStudio 0x102c07b80 main + 172 34 dyld 0x1a65cbe50 start + 2544
> 
> Thread 1: 0 libsystem_pthread.dylib 0x1a68f0e18 start_wqthread + 0
> 
> Thread 2: 0 libsystem_pthread.dylib 0x1a68f0e18 start_wqthread + 0
> 
> Thread 3: 0 libsystem_pthread.dylib 0x1a68f0e18 start_wqthread + 0
> 
> Thread 4:: com.apple.NSEventThread 0 libsystem_kernel.dylib 0x1a68b6df0 mach_msg2_trap + 8 1 libsystem_kernel.dylib 0x1a68c88d8 mach_msg2_internal + 80 2 libsystem_kernel.dylib 0x1a68bf638 mach_msg_overwrite + 540 3 libsystem_kernel.dylib 0x1a68b716c mach_msg + 24 4 CoreFoundation 0x1a69d5bdc __CFRunLoopServiceMachPort + 160 5 CoreFoundation 0x1a69d44c8 __CFRunLoopRun + 1232 6 CoreFoundation 0x1a69d38a4 CFRunLoopRunSpecific + 612 7 AppKit 0x1a9d56248 _NSEventThread + 172 8 libsystem_pthread.dylib 0x1a68f606c _pthread_start + 148 9 libsystem_pthread.dylib 0x1a68f0e2c thread_start + 8
> 
> Thread 5:: iotc_conn_hdlr 0 libsystem_kernel.dylib 0x1a68c1a74 __select + 8 1 libBambuSource.dylib 0x128f684d8 connectionHandlerThread + 472 2 libsystem_pthread.dylib 0x1a68f606c _pthread_start + 148 3 libsystem_pthread.dylib 0x1a68f0e2c thread_start + 8
> 
> Thread 6:: iotc_taskmgr 0 libsystem_kernel.dylib 0x1a68ba3e0 __semwait_signal + 8 1 libsystem_c.dylib 0x1a67c45b0 nanosleep + 220 2 libsystem_c.dylib 0x1a67c44c8 usleep + 68 3 libBambuSource.dylib 0x128f67400 tutk_TaskMng_Run + 288 4 libsystem_pthread.dylib 0x1a68f606c _pthread_start + 148 5 libsystem_pthread.dylib 0x1a68f0e2c thread_start + 8
> 
[... cut ...] ub Controller Thunderbolt Bus: Mac mini, Apple Inc. Thunderbolt Bus: Mac mini, Apple Inc.
> 
> **Desktop (please complete the following information):**
> 
> * OS: MAC OS Ventura
> * M1 Chip

@bravojohnny1 
Is this issue 100% reproducible?
and have you installed multiple version of BambuStudio or run from different dmg at the same time?

for reinstall, you can try following steps:
delete 'BambuStudio' in the Applications (Move to Trash)
rm /Users/user.name/Library/Application\ Support/BambuStudio

bravojohnny1, 2023-02-11:
The latest Version was working fine. Sometimes last week it crashed. Since then it is 100% reproducible. 
I installed V1.04.0.xxx and the behaviour is the same. 
Tried reinstalling, deleting etc. multiple times.

SaltWei, 2023-03-02:
Is this issue 100% reproducible with the latest v1.5.0 public beta?

lanewei120, 2023-03-16:
it is a issues caused by previous data not cleaned
next time when this issue happens, we can clean the application by following commands
rm -rf /Applications/BambuStudio.app
then we can rename the directory /Users/user.name/Library/Application\ Support/BambuStudio to a backup one
and packed this directory to us for a analysis

amfdesign, 2023-04-01:
Ich habe das gleiche Problem, leider kann ich diesen Ordner nicht finden /Library/Application\ Support/**_BambuStudio_**

amfdesign, 2023-04-01:
I have the same problem, I can't find this folder /Library/Application\ Support/BambuStudio

lanewei120, 2023-04-02:
> I have the same problem, I can't find this folder /Library/Application\ Support/BambuStudio

it is /Applications/BambuStudio.app
and another directory should use your user name instead
/Users/**user.name**/Library/Application\ Support/BambuStudio

lanewei120, 2023-04-18:
add a solution in v1.6 to postpone opengl init when the window is not ready
it can improve this behavior
close this issue currently
if still has this issue, let's reopen it and dig into it

closed 2023-04-18 by lanewei120