bambulab/BambuStudio #1272: Crash when tree support slim extends over bed and "on build plate only" is turned on

SavexG, 2023-02-10:
I'm on Linux (EndevourOS) using this appimage:

https://www.spui.uk/BambuStudio/BambuStudio_F37_v01.04.02.13-14-gd1b93c16.AppImage

_Originally posted by @spuiuk in https://github.com/bambulab/BambuStudio/issues/760#issuecomment-1415535654_

When trying to slice the pegboard modplate for the p1p on plate 6 the tree supports extend over the printing bed. 
When I turn on the setting "on build plate only" and then try to slice it again bambustudio crashes.

settings I changed from the default: 
support (turned on)
type (Tree)
Style (Tree Slim)
build on plate only (turned on)
support critical regions only (turned on)

tangketan, 2023-02-12:
That's a known bug and has been fixed. Please wait for the next release.

spuiuk, 2023-02-13:
@tangketan is the patch in the v1.4.2 branch? I could build the latest Appimage with the latest patches for some user testing.

spuiuk, 2023-02-13:
I have compiled a new Appimage this morning which is available at
http://www.spui.uk/BambuStudio/BambuStudio_F37_v01.04.02.13-17-g9fddc497.AppImage

This only contains the following patches above what you have in the Appimage you currently use. None of these seem to be related to this issue though
9fddc497 (FIX: fix crash when switch printer, 2023-02-07)
079d41f9 (FIX: use shmget according to bambu_source, 2023-02-02)
e89f535f (FIX: task thumbnail state update, 2023-02-06)

tangketan, 2023-08-02:
Sorry for the late reply. The latest Studio 1.7 should have solved the bug. Please take a look.

closed 2023-08-02 by tangketan