bambulab/BambuStudio #1273: [Issue] Saving file causes duplicates

BambuLabDiscord, 2023-02-11:
Not sure how to repro, other than saving, but I find that when I save a 3mf and reload it, it will duplicate all models (in the exact same position).

lanewei120, 2023-02-11:
@BambuLabDiscord 
thanks for reporting this issue
this is a duplicate issue of https://github.com/bambulab/BambuStudio/issues/1185
we have fixed it locally and will be in next version

For these project which loaded duplicated
We are very sorry that you need to remove the duplicated model manually

BTW: I guess when you click on the project in the home page
there is no action because of the bug in Studio
and then when you click again the model loaded with duplicate ones

lanewei120, 2023-03-01:
already in v1.5 just released

closed 2023-03-01 by lanewei120