bambulab/BambuStudio #1275: Bambu Studio - Linux - Cannot send to print

candreacchio, 2023-02-12:
**Describe the bug**
After slicing, and pressing the send to print option, a window appears, then when sending the print it comes up with an error " Exception: std::bad_alloc

**To Reproduce**
Steps to reproduce the behavior:
1. Open Bambu studio on linux (latest github code)
2. slice any model
3. send to print
4. hit send
5. Crash

**Expected behavior**
Bambu Studio does not crash

**Screenshots**
[image]

**Desktop (please complete the following information):**
 - OS: Kubunutu 22.10

spuiuk, 2023-02-13:
This has already been fixed
Please take a look at the comment 
https://github.com/bambulab/BambuStudio/issues/760#issuecomment-1398086890

spuiuk, 2023-02-13:
I think the fix is in the dev branch though ie. v1.4.2 and haven't been merged with the master branch yet.

lanewei120, 2023-02-15:
yes, @spuiuk  is correct

please use branch v1.4.2 currently

deftdawg, 2023-02-23:
Hopefully the fix gets merged soon.

@zhaofengli's [NixOS Bambu Studio package repo](https://github.com/zhaofengli/nixpkgs/tree/bambu-studio) is also impacted by this bug.

lanewei120, 2023-02-24:
already merged to master
now master should work fine now

lanewei120, 2023-03-01:
already in v1.5 just released

closed 2023-03-01 by lanewei120