bambulab/BambuStudio #1276: Function to name / label each build plate

Learoy1, 2023-02-12:
I have projects in Bambu studio with a lot of different parts on different plates, currently when you send the print to the printer it names it  something like 'plate 2' but when I go back to reprint from history it's sometimes difficult to know which part is which. I need to more easily identify  parts. Can we have a function where we can name each build plate in the slicer. (For example left Midsole, right Midsole, left upper , right sole, etc)

mrhasbean, 2023-02-18:
Came to add this request too. Definitely a useful feature for managing parts of a (large) project or where elements are similar on some plates.

RyanEwen, 2023-03-20:
This drives me crazy - you can rename the plate in the little window that appears when sending it, but it doesn't help

RyanEwen, 2023-03-29:
I think I just realized the issue - if you rename the plate when Sending or Printing, you must press Enter in the text field, otherwise the changed name doesn't get used. It took me 3 months to realize I was pressing Enter sometimes and not other times.

StoneLiBambulab, 2023-04-26:
plan to add it in version 1.7

Thanks!