bambulab/BambuStudio #1277: User defined Filament Preset can't be used in AMS

Laserstrahl, 2023-02-12:
**Describe the bug**
When you create a new preset (user defined preset) for a filament it is saved and can be accessed from the drop-down menu under "prepare" tab. When you add an AMS to the system, go to "device" tab, load a filament into the AMS, and click the Icon under the "K" to open up the select filament menu. There you can't see custom filaments, only the system pre-defined ones.

**3mf File for This Bug**
Doesn't matter which 3mf file

**To Reproduce**
see above

**Expected behavior**
For my understanding the user defined filament presets should be available within the AMS Menu to load a user preset to a filament slot of the AMS.

**Screenshots**
Showing user-defined presets
[image]
Not showing up in AMS Filament selection menu
[image]

**Desktop (please complete the following information):**
 - OS: Windows 10
 - BambuStudio v. 1.4.2.13

SaltWei, 2023-02-13:
@bambu123 Please help to check whether this is a bug. I think this is a feature request, rather than bug, right?

bambu123, 2023-02-13:
Not support custom filament id in AMS.
And not support setting id in AMS.
Here the filament id is different from the setting id.
Custom presets add a setting id, not a filament id.

By the way:
Set to AMS has less effect than selecting in preparing the preset.
So what do you think about respecting?

BastiPy, 2023-02-13:
I would like to add that it is also not possible to use the Dynamic Flow Calibration with User Defined Presets.  @bambu123 @StoneLiBambulab Should I open a new Issue for this?

Laserstrahl, 2023-02-17:
> Not support custom filament id in AMS. And not support setting id in AMS. Here the filament id is different from the setting id. Custom presets add a setting id, not a filament id.
> 
> By the way: Set to AMS has less effect than selecting in preparing the preset. So what do you think about respecting?

Anyway also If I use BambuLab Original Filament (which is impossible to order in europe currently) it is not taking the changes over. Please fix this bug in on of the next releases. :)

bambu123, 2023-02-20:
Dynamic Flow Calibration may be another issue, I'll confirm it.

zviratko, 2023-02-21:
@bambu123 > So what do you think about respecting?

i.e. When using a filament with very low temperature requirements (Fillamentum NonOilen is what I used that prints between 175-190ºC and must not be heated above 210ºC), not having "custom" material in the AMS/Printer itself still uses the generic PLA temperatures that might be too high.

There are other materials, where setting temperatures used for loading/purging or other operations might be needed. It would be great to be able to import temperatures and other properties from the slicer and have the printer respect them.

StoneLiBambulab, 2023-04-24:
included in version 1.6, Custom filament presets should be selectable in Dynamic Flow Calibration.
Closed.

Stone Li

closed 2023-04-24 by StoneLiBambulab

ArnonAfriat, 2023-12-10:
@StoneLiBambulab 
This issue still reproduces in Bambulab 1.7.7.89 on Windows.