bambulab/BambuStudio #1278: Increase Probe Grid Size

chrishilbert, 2023-02-12:
**Is your feature request related to a problem? Please describe.**
There are still places on the bed that are too low or too high when printing.

**Describe the solution you'd like**
Allow an option to increase/decrease the number of bed leveling probe points in the grid.  4x4, 5x5, 6x6, 7x7, etc.

**Describe alternatives you've considered**
N/A

**Additional context**

SaltWei, 2023-02-13:
This is not a slicer feature request. Should be raised to printer firmware.

chrishilbert, 2023-02-13:
Where is the printer firmware Git?

SaltWei, 2023-02-14:
Thank you very much for suggestions!

I have transfered this ticket to printer team.

For issue about firmware and printer, please move to https://forum.bambulab.com/ or BanbuHandy APP to report ticket (https://wiki.bambulab.com/en/x1/troubleshooting/upload-log) for better support. Thanks!