bambulab/BambuStudio #1285: Failed uploading print file

Ali229, 2023-02-15:
**Describe the bug**
Cannot send print to SD card. This has been a problem for months, then finally the manually defining the printer's IP address feature was introduced inside bambu studio since then I had no problem, but since 2-3 days ago this issue is back. I do not have any kind of VPN or antivirus.

**To Reproduce**
1. Slice a file
2. From the top right drop down select "send"
3. Confirm Send
4. See error "Failed uploading print file"

**Expected behavior**
Send the file to the printer's SD card

**Screenshots**
[image]

**Desktop (please complete the following information):**
 - OS: Windows
 - Version 11

StoneLiBambulab, 2023-02-16:
Hi @Ali229 ,
maybe the sd card is full, and could help to upload the log files zip, please refs to 
https://wiki.bambulab.com/en/x1/troubleshooting/bambu-studio-log

Thanks!

Stone Li

Ali229, 2023-02-16:
@StoneLiBambulab Thank you for the reply, I have already formatted the card and even tried a different card in case of corruption, by the way I can still print from the same SD card by manually placing files in it from the computer. I've also attached the logs here. Looking forward to getting this resolved, let me know if you need anything else.
[debug_Wed_Feb_15_20_38_36_18136.log.zip](https://github.com/bambulab/BambuStudio/files/10751097/debug_Wed_Feb_15_20_38_36_18136.log.zip)

walterwongbbl, 2023-02-16:
> @StoneLiBambulab Thank you for the reply, I have already formatted the card and even tried a different card in case of corruption, by the way I can still print from the same SD card by manually [...]

Hi.
what is the firmware version of your printer?
please try upgrading to the latest version and try again.

Ali229, 2023-02-16:
I did have update the firmware the other day, I think it happened after the update but I'm not 100% on this statement (2/6 is when I last uploaded a file on the card, 2/7 is when the firmware was released).

Here is what I have:
**Bambu Lab X1-Carbon Firmware: 01.04.00.00 
AMS 1 Firmware: 00.00.05.96
Bambu Studio Version: 1.4.2.13**

Ali229, 2023-02-16:
Here is a gif of it happening
[image]

walterwongbbl, 2023-02-16:
> 

Thank you very much.
We have seen LOG. This is an OpenSSL error. It seems to be related to the printer firmware. Please try unbind the printer and then bind again.

Ali229, 2023-02-16:
Rebinding didn't help, here are the steps I followed:
1. Unbind the printer from LCD
2. Bind from bambu handy using the QR code on LCD
3. Restart both bambu studio and the printer
4. Still seeing same error while sending prints to sd card

Also attached logs again
[debug_Wed_Feb_15_22_49_59_21932.log.zip](https://github.com/bambulab/BambuStudio/files/10751991/debug_Wed_Feb_15_22_49_59_21932.log.zip)

Ali229, 2023-02-16:
In a desperate attempt to try everything I could, I tried unbinding both the slicer and printer, didn't work, I then FTP'd into the printer using WinSCP successfully and uploading to the SD card. I figured nothing is wrong with the printer or the SD card. So I reinstalled the slicer completely (deleting user data) and I was able to send to SD card again. @Absolo13 Please try this and report back if you're able to resolve it as well.

Absolo13, 2023-02-16:
@Ali229  I reinstalled bambu studio (deleting user data) but it still doesn't work :(
Sd card is good, i see my machine and the webcam with bambu studio all is ok except this failed uploading print file :(

Ali229, 2023-02-16:
I would try the combination of the things I tried(just to be sure the culprit isn't one of these):
1. Unbind printer from screen and then from slicer
2. Reset printer to factory default from screen
3. Uninstall slicer and files in configuration folder
4. Reinstall slicer
5. Restart your router(s)

Absolo13, 2023-02-16:
I'll try but when you say "Unbind printer from screen" do  you mean disconnect the account from the screen ?
Thanks

Ali229, 2023-02-16:
Yes

krsteon, 2023-02-17:
Running into this issue as well on both Windows 10 and Windows 11 PCs.  On V 1.4.2.13 in Bambu Studio and 01.04.00.00 Firmware on the X1C.  Any fix found?

walterwongbbl, 2023-02-17:
> I would try the combination of the things I tried(just to be sure the culprit isn't one of these):
> 
> 1. Unbind printer from screen and then from slicer
> 2. Reset printer to factory default from screen
> 3. Uninstall slicer and delete files in configuration folder
> 4. Reinstall slicer
> 5. Restart your router(s)

Thank you very much.
can u privide us BambuStudio.conf file. we will get more information.
[image]
[image]

walterwongbbl, 2023-02-17:
> Running into this issue as well on both Windows 10 and Windows 11 PCs. On V 1.4.2.13 in Bambu Studio and 01.04.00.00 Firmware on the X1C. Any fix found?

Hi.
we are trying to reproduce this issue. 
You can try to logout the printer and then login the printer again. 
[image]
[image]

p2im0, 2023-02-18:
I am also experiencing the same issue.  Tried unbinding printer, slicer, phone, and logging back in and binding.  I can start prints fine from the slicer but cannot send files to the SD.
The only thing I know I did earlier today was rotate the "LAN Only" code on the Printer's Network Settings page (by accident).  Not sure why that would have to do with it, but I do remember getting a prompt for the IP and that code the first time I attempted to send a file to the MicroSD in my printer and I don't know how to make it re-prompt me.

Ali229, 2023-02-18:
I'm reopening this issue as other people have the same problem!
@p2im0 did you also try restarting your network?

p2im0, 2023-02-18:
> I'm reopening this issue as other people have the same problem! @p2im0 did you also try restarting your network?

Apologies I missed the step of re-installing the slicer **and** deleting user data!  After slicer reinstall and deleting the config files that did fix the issue!  Much appreciated for the Ludicrous speedy response :)

vjenne, 2023-02-19:
Yes i have the same problem too!

RonnieBlaze, 2023-02-20:
I started to have issues sending to the printer as well.  I either get the same error as above, or one that says

Failed uploading print file[Unknown Error.status=150, body=failed sending data to the printer.

And if i go to the printer there is a blank file named with the job i was trying to send. 
I have uninstalled Bambu studio and deleted user files, logged the printer out and back in, and it was only able to send 1 file before it stops sending.

Absolo13, 2023-02-21:
I have exactly the same problem but I noticed that on the other hand the "Print plate" button of Bambu Studio works and sends the print to the printer. Is it the same for you?

Absolo13, 2023-02-21:
I solved the problem thanks to the support of Bambu Labs.
Follow these steps in order:
1- Disconnect your user account from the printer
2- Disconnect your bambu studio user account (on your PC)
3- Reconnect your printer with your user account (thanks to your smartphone)
4- Completely uninstall bambu studio from your pc
5- Reinstall bambu studio on your PC and log in to your account.

It worked for me.

RonnieBlaze, 2023-02-21:
> I solved the problem thanks to the support of Bambu Labs. Follow these steps in order: 1- Disconnect your user account from the printer 2- Disconnect your bambu studio user account (on your PC) 3- [...]

So far this is working, but this is almost the same thing i tried once before and it worked for a couple of files and then stopped.. will see if they stays working or not..

krsteon, 2023-02-22:
> 

worked for me as well

Ali229, 2023-02-26:
Unfortunately this error has started reoccurring this needs to be fixed, can't keep relinking again and again...
[image]

walterwongbbl, 2023-02-27:
> 不幸的是，此错误已开始再次发生，这需要修复，无法一次又一次地重新链接...... [image]

> Unfortunately this error has started reoccurring this needs to be fixed, can't keep relinking again and again... [image]

Hi.
In the next version, we will check the network status before sending.
and allow users to manually enter the IP and access code of the printer.

RonnieBlaze, 2023-03-06:
well my issue came back, like i thought it would.. i even tried updating to the 1.05.00.59 version and same thing.

walterwongbbl, 2023-03-06:
> well my issue came back, like i thought it would.. i even tried updating to the 1.05.00.59 version and same thing.

Hi. 
Did you uninstall Studio first? 
it is best to uninstall slicer and delete files in configuration folder before installing the new version.

RonnieBlaze, 2023-03-06:
> > well my issue came back, like i thought it would.. i even tried updating to the 1.05.00.59 version and same thing.
> 
> Hi. Did you uninstall Studio first? it is best to uninstall slicer and delete files in configuration folder before installing the new version.

ya it only seems to work for a couple of times.. and then it goes back to not working.. My printer is set with a static ip address from my firewall so its never going to change..

petenun, 2023-04-01:
Still having this problem with latest firmware and software as of 4/1/2023. Tried jumping through the above hoops with only temporary success. Printer has a reserved IP address via router and never changes (static addressing would be welcome on the printer, hint, hint.) Software has no problems sending print job to the printer, so it's obviously communicating; it just can't send to the SD card. New, newly formatted SD card (128gb) has the same problem.

[image]

walterwongbbl, 2023-04-03:
> Still having this problem with latest firmware (01.04.01.00) and software (1.5.0.61) . Tried jumping through the above hoops with only temporary success. Printer has a reserved IP address via router and never changes (static addressing would be welcome on the printer, hint, hint.) Software has no problems sending print job to the printer, so it's obviously communicating; it just can't send to the SD card. New, newly formatted SD card (128gb) has the same problem. This is occurring from four different Windows computers, each running Windows 10 Pro 22H2, and each with a unique instance of Bambu Studio installed, all connected to the same printer.
> 
> 1.5.0.61
> 
> [image]
> 
> I also ran a network test via the bambu Studio Help menu. Note that the Test Storage Upload test succeeded, even though it fails in real life.
> 
> [image]

Hi.
This network test does not test the status of ftp.Include only the connection status to the server.
Have you checked the status of your router?
We have encountered some users using very complex network environments, resulting in FTP work failures.
In 95% of cases, by reinstalling the bambu studio software and rebinding the printer, the problem of sending files to the printer's SD card failing can be resolved.

petenun, 2023-04-03:
> 

I do have a couple questions:
1. If the "Test Storage Upload" does not test upload to storage, what does it test?
2. For my understanding, what path does the FPT take? Does it FTP from each workstation directly to the local address of the printer? Or does it FTP to "the cloud", when then transfers the file to the printer? This makes a big difference if troubleshooting FTP connectivity, whether internal or external.
3. Is it at all feasible that the same feature (upload to SD card) would break on ALL FOUR of my Windows workstations at the same time?
4. To answer your question, my network configuration is relatively straight forward. I have a single subnet with no ports blocked internally or outbound to the internet. Inbound from the internet is a different story, with most ports blocked. 

Thanks,
Pete

walterwongbbl, 2023-08-02:
Hi.
Please use the latest version of Bambu Studio

RonnieBlaze, 2023-08-03:
> Hi,
> 
> still running into the same issue with the new version 1.7.1.62. Pops the dialog and asks me to enter the IP-address and access-code.
> 
> Printer is definity online and the code is correct (i can turn live camera view on, so i guess everythin is setup correctly).

as stupid as this sounds, try rebooting your computer.. every time i have issues sending a file. if i reboot my computer it seems to work for awhile.

JR250, 2023-12-02:
Hi all. FWIW, I just came across this issue today while printing a part I downloaded. I tried resetting, reinstalling, etc and nothing worked. I then tried a different part and it worked fine. This lead me to believe either credentials, machine info, or setup was in that file from the original author.

To be clear, the part loaded and sliced fine in Studio. Only trying to send it to the printer failed with the "Failed uploading" error.

Solution: extract and save the .STL, and build the project from scratch.

If either credentials or specific machine binding are indeed included in 3mf projects files, and I'm speculating here, perhaps there should be an option to exclude that when loading a project.

maruel, 2023-12-18:
Confirming @JR250 's observation. In both case dropping the 3mf and restarting from the base STL model fixed the problem.

walterwongbbl, 2023-12-18:
If you have any other questions, please create a new theme

closed 2023-12-18 by walterwongbbl

maruel, 2023-12-20:
@walterwongbbl I think the observation is: there are 3mf fles that bambulab will process and generate an invalid gcode that the printer will refuse. The error message is confusing and generic. I think the action item would be to improve the error message, or fix bambu studio to not generate invalid gcode.

ioscanner, 2024-01-08:
I am seeing this issue too.  This issue should get fixed!  It is clear data is in the 3mf files that shouldn't be.  If that kind of information is left in the 3mf files that could be a security issue!  I do plan to do a penetration test on downloading projects and printing.  My guess is someone never even thought about clearing the project files first or even testing that malicious content is not embedded. I am sure bad things can be performed on the computer running Bambu studio as well as the printer.

Bambu: Have you performed a security audit of the Maker World process tied into Bambu?  This happens to be what I do for a living.

StephanBieker, 2024-01-16:
> Hi. In the next version, we will check the network status before sending. and allow users to manually enter the IP and access code of the printer. this will improve the success rate of sending

This obviously didn't work. This annoying bug exists until today. It works for some times ojn all devices and then it stops working and your clueless test appears. This ip test didnt fix it in one time.

please fix this very annoying bug soon.

gothickitty93, 2024-03-13:
> Still having this problem with latest firmware (01.04.01.00) and software (1.5.0.61) . Tried jumping through the above hoops with only temporary success. Printer has a reserved IP address via router [...]

Where do I find this UI? I have no idea where this dialog box is in either Bambu Studio or OrcaSlicer, and it would really help me out as I'm trying to connect to my printer via my personal VPN

MrStefaan, 2024-09-07:
After TWENTY months it's still not fixed, it's a critical bug making it impossible to use our printers in lan mode, what the hell are you doing???

Nico04, 2024-09-27:
Having same issue here, it should be opened again no ?

jeremyg58, 2024-11-15:
The fact that this bug still exists is absolutely insane. I can uninstall, reinstall, link, unlink, bind, unbind, any combination of these, and I absolutely cannot send files to the printer. Period. It has never worked once. I've used multiple computers, updated everything.

An absolute joke. A critical function that hasn't been resolved after nearly 2 YEARS.

And now it seems like BBL isn't even responding to this issue anymore???

Seriously considering getting rid of this paperweight if this isn't resolved.

StephanBieker, 2024-11-15:
In my times, sometimes it works once and then it doesn't. I cannot figure
it out. It drives me nuts.

Nico04, 2024-11-16:
I'm in contact with the support for weeks now, trying to find a solution: 
I've tried to format the SD card using the printer (again), unlink, link, reset, reinstall every bambulab related software or firmware.
At the end they send me a screen & screen cable replacement, but I still have the same issue 👎