bambulab/BambuStudio #1286: Wrong layer count on print completion

bolsoncerrado, 2023-02-15:
**Describe the bug**
A print job just completed but the layer count seems incorrect.

**To Reproduce**
Steps to reproduce the behavior:
1. Launch a print from BStudio
2. Let it print.
3. From BStudio, once the job is finished, layer count is incorrectly displayed.

**Expected behavior**
To display correct layer count.

**Screenshots**
[image]

**Desktop (please complete the following information):**
 - OS: MacOS Ventura
 - Mac M1
 - BStudio for Mac 1.4.2.xxx

SaltWei, 2023-02-16:
This is a known issue of studio and have been fixed in internal version.
The user version will be released soon.

cameo69, 2023-02-26:
Will the fix also include the case that the actual count is greater than the total count?
[image]

SaltWei, 2023-02-27:
> Will the fix also include the case that the actual count is greater than the total count?

We only met the case the actual count is smaller than the total count.
Could you share this case by 3MF file to us? We will have a test and check.
Thank you very much!

SaltWei, 2023-03-01:
Included in v1.5.0 Public Beta

closed 2023-03-01 by SaltWei