bambulab/BambuStudio #1287: Can not print special part

theogott, 2023-02-16:
**Describe the bug**
I can print other parts, but this part always runs into Problems.

**3mf File for This Bug**
It does not matter which settings or supports i tried. It never prints succesfully.

**To Reproduce**
Steps to reproduce the behavior:
Just try to print the attached part in any way.

**Expected behavior**
It should print especially with all sorts of support enabled. In tree support i used even 3 layers for the support.
I tried with only 1 Filament and also tried using Support W Filament for the Interface.
I tried with PLA and also PETG, i always run into problems.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Desktop (please complete the following information):**
 - OS: [e.g. iOS, Windows]
 - Version [e.g. 22]
[Combined Special C.zip](https://github.com/bambulab/BambuStudio/files/10753213/Combined.Special.C.zip)

SaltWei, 2023-02-21:
@MusangKingBambulab The model is not supported well. Please have a check

MusangKingBambulab, 2023-04-10:
Hi, @theogott. I have checked your 3mf and found the issue. The printing result illustrates that the adhesion area between the first layer and the build plate is too small but with a large build height. That may cause unstable printing like some parts came off and spaghetti. I suggest changing the Support Threshold angle to 50 which will generate more support to improve the stability.
[image]

theogott, 2023-04-10:
My suggestion is, if the Slicer can do such a suggestion automatically, because the average User will never find out that he needs to change exactly this value.
Having said that we can close the case. But i have another open case with horzontal Lines, here:

Ticket No: US230331263001 * Print surface is bad(spot/seam/etc)
00M00A270401037 * 23/03/31 08:21
https://bambulab.com/en/my/support/tickets

Maybe you can take a look on this.

MusangKingBambulab, 2023-04-12:
@theogott Thanks for your suggestion. We will add tips to remind user aiming for such cases in the future release as soon as possible.

As for your ticket, we will reply and provide solutions soon. Thanks again.

MusangKingBambulab, 2023-04-13:
@theogott I have printed your uploaded 3mf and I can not reproduce your issue.

theogott, 2023-04-13:
The we can close this ticket. Lets take a look on the "striped Buddha".