bambulab/BambuStudio #1289: Spiral Vase Mode checkbox crashes Bambu Studio on macOS

Manpoozle, 2023-02-16:
-------------------------------------
Translated Report (Full Report Below)
-------------------------------------

Process:               BambuStudio [23499]
Path:                  /Applications/BambuStudio.app/Contents/MacOS/BambuStudio
Identifier:            com.bambulab.bambu-studio
Version:               01.04.02.13 ()
Code Type:             X86-64 (Native)
Parent Process:        launchd [1]
User ID:               501

Date/Time:             2023-02-16 15:49:33.0662 -0800
OS Version:            macOS 12.6.2 (21G320)
Report Version:        12
Anonymous UUID:        C4E6958B-6520-F4CB-E199-DFA9912D220D

Sleep/Wake UUID:       75A9E4E6-1696-41D7-BBAE-C32732EEAC1B

Time Awake Since Boot: 840000 seconds
Time Since Wake:       845217 seconds

System Integrity Protection: enabled

Crashed Thread:        0  bambustu_main  Dispatch queue: com.apple.main-thread

Exception Type:        EXC_CRASH (SIGABRT)
Exception Codes:       0x0000000000000000, 0x0000000000000000
Exception Note:        EXC_CORPSE_NOTIFY

Application Specific Information:
abort() called

Thread 0 Crashed:: bambustu_main Dispatch queue: com.apple.main-thread
0   libsystem_kernel.dylib        	    0x7ff81d70700e __pthread_kill + 10
1   libsystem_pthread.dylib       	    0x7ff81d73d1ff pthread_kill + 263
2   libsystem_c.dylib             	    0x7ff81d688d24 abort + 123
3   BambuStudio                   	       0x10c4d1ab9 wxAbort() + 9
4   BambuStudio                   	       0x10c578f6a wxEvtHandler::WXConsumeException() + 154
5   BambuStudio                   	       0x10c578ea0 wxEvtHandler::SafelyProcessEvent(wxEvent&) + 32
6   BambuStudio                   	       0x10c454914 wxToggleButton::OSXHandleClicked(double) + 132
7   AppKit                        	    0x7ff8204400ce -[NSApplication(NSResponder) sendAction:to:from:] + 288
8   AppKit                        	    0x7ff82043ff74 -[NSControl sendAction:to:] + 86
9   AppKit                        	    0x7ff82043fea6 __26-[NSCell _sendActionFrom:]_block_invoke + 131
10  AppKit                        	    0x7ff82043fdaf -[NSCell _sendActionFrom:] + 171
11  AppKit                        	    0x7ff82043fcf7 -[NSButtonCell _sendActionFrom:] + 96
12  AppKit                        	    0x7ff82043cb6f NSControlTrackMouse + 1813
13  AppKit                        	    0x7ff82043c436 -[NSCell trackMouse:inRect:ofView:untilMouseUp:] + 121
14  AppKit                        	    0x7ff82043c306 -[NSButtonCell trackMouse:inRect:ofView:untilMouseUp:] + 679
15  AppKit                        	    0x7ff82043b6d6 -[NSControl mouseDown:] + 678
16  BambuStudio                   	       0x10c4b38ce wxWidgetCocoaImpl::mouseEvent(NSEvent*, NSView*, void*) + 318
17  AppKit                        	    0x7ff820439bc1 -[NSWindow(NSEventRouting) _handleMouseDownEvent:isDelayedEvent:] + 4859
18  AppKit                        	    0x7ff8203add7e -[NSWindow(NSEventRouting) _reallySendEvent:isDelayedEvent:] + 2582
19  AppKit                        	    0x7ff8203ad14e -[NSWindow(NSEventRouting) sendEvent:] + 352
20  BambuStudio                   	       0x10c4a01db -[wxNSWindow sendEvent:] + 123
21  AppKit                        	    0x7ff8203ab524 -[NSApplication(NSEvent) sendEvent:] + 352
22  BambuStudio                   	       0x10c41e312 -[wxNSApplication sendEvent:] + 98
23  AppKit                        	    0x7ff82066418b -[NSApplication _handleEvent:] + 65
24  AppKit                        	    0x7ff82022cd3e -[NSApplication run] + 623
25  BambuStudio                   	       0x10c496cae wxGUIEventLoop::OSXDoRun() + 174
26  BambuStudio                   	       0x10c58355d wxCFEventLoop::DoRun() + 29
27  BambuStudio                   	       0x10c4f7eb5 wxEventLoopBase::Run() + 85
28  BambuStudio                   	       0x10c4d03c4 wxAppConsoleBase::MainLoop() + 116
29  BambuStudio                   	       0x10c458d3a wxApp::OnRun() + 26
30  BambuStudio                   	       0x10c52bc93 wxEntry(int&, wchar_t**) + 131
31  BambuStudio                   	       0x10a39f837 Slic3r::GUI::GUI_Run(Slic3r::GUI::GUI_InitParams&) + 167
32  BambuStudio                   	       0x109857b14 Slic3r::CLI::run(int, char**) + 18340
33  BambuStudio                   	       0x1098725d0 main + 256
34  dyld                          	       0x11af6d52e start + 462

[... 439 lines cut ...]
}
}

Model: iMac17,1, BootROM 170.0.0.0.0, 4 processors, Quad-Core Intel Core i7, 4 GHz, 32 GB, SMC 2.34f3
Graphics: AMD Radeon R9 M395X, AMD Radeon R9 M395X, PCIe, 4 GB
Display: iMac, Retina 5K (5120 x 2880), Main, MirrorOff, Online
Memory Module: BANK 0/DIMM0, 8 GB, DDR3, 1867 MHz, 0x80AD, 0x484D54343147533642465238412D52442020
Memory Module: BANK 0/DIMM1, 8 GB, DDR3, 1867 MHz, 0x80AD, 0x484D54343147533642465238412D52442020
Memory Module: BANK 1/DIMM0, 8 GB, DDR3, 1867 MHz, 0x80AD, 0x484D54343147533642465238412D52442020
Memory Module: BANK 1/DIMM1, 8 GB, DDR3, 1867 MHz, 0x80AD, 0x484D54343147533642465238412D52442020
AirPort: spairport_wireless_card_type_wifi (0x14E4, 0x14A), Broadcom BCM43xx 1.0 (7.77.111.1 AirPortDriverBrcmNIC-1710.4)
AirPort: 
Bluetooth: Version (null), 0 services, 0 devices, 0 incoming serial ports
Network Service: Wi-Fi, AirPort, en1
Serial ATA Device: APPLE SSD SM1024G, 1 TB
USB Device: USB30Bus
USB Device: USB3.0 Hub
USB Device: SDDR-B531
USB Device: Bluetooth USB Host Controller
USB Device: FaceTime HD Camera (Built-in)
USB Device: USB2.0 Hub
USB Device: Magic Keyboard
USB Device: Audioengine 2+
USB Device: Yeti Stereo Microphone
Thunderbolt Bus: iMac, Apple Inc., 28.1

bambu123, 2023-02-22:
Can you tell me how to reproduce it?
Simply clicking the check box works well.

Manpoozle, 2023-02-22:
Can I get you a dump of my Printer, Filament, and Process settings are? It readily reproduces for me with no model loaded opening here, clicking the Others tab, checking the box, and interestingly doesn't matter which option you select, both crash immediately.

[image]

bambu123, 2023-02-22:
Yes, it maybe relates to the presets you have edited and selected. 
Can you zip and send me your configuration folder?
I need the 'system' and 'user' subfolders in it which contain all presets.

Manpoozle, 2023-02-22:
[BambuStudio.zip](https://github.com/bambulab/BambuStudio/files/10799368/BambuStudio.zip)

bambu123, 2023-02-22:
I got the issue. I'm sorry.
It's a known issue when setting spiral mode while wipe_tower is disabled.
The issue will be fixed in the coming version 1.5 at the end of Feb.

Manpoozle, 2023-02-22:
Thanks @bambu123. I really appreciate you having a look.

closed 2023-03-08 by bambu123