bambulab/BambuStudio #1290: Additional speed profiles and ability to select BEFORE printing

Somerandomgithugguy, 2023-02-17:
Dear BL devs,

Please add the following features:

1) 25% and 75% speed profiles
2) A user-definable speed of 0-100%.
3) The ability to define the above "quick speed" profile for the print BEFORE printing (currently we can only do this after the print has already started).

Many thanks.

julie777, 2023-02-22:
I would like an even more specific improvement. I would like to select a speed to start the print and a speed to use after a firm foundation is created. I don't know if the split point should be fixed, or set by height or layers. I do know that starting at a slow speed and getting the first few layers down before ramping it up to sport works well for me, but I have to change it manually at the moment.

Somerandomgithugguy, 2023-04-17:
Hi BL Devs,

The community really wants to see this so I hope you can get around to doing it. It would be a big quality of life improvement. :)

Regards,

R.

daliudzius, 2023-05-03:
> 3. The ability to define the above "quick speed" profile for the print BEFORE printing (currently we can only do this after the print has already started).

**This would be a huge quality of life improvement**. I primarily print in "silent" speed mode since it prints loud by default, and it's a hassle to wait for prints to start just to change the speed. Having silent mode change chamber fan speeds to like 50-70% is a separate issue but would also be a huge improvement.

Somerandomgithugguy, 2023-05-04:
> Having selectable speeds during a print is very handy for fine tuning (continuously variable would be best), but if you know what you want ahead of time, why not just set the speeds, accelerations, fan speeds, etc. exactly as fast as you want them for a particular situation? You can save the new profile(s) with "Silent" or whatever in the name and easily switch between them based on your needs.

We know how to go in and manually set ever speed setting but those workarounds are obviously not what we are asking for. We are asking for a slider feature that does it all easily and conveniently without needing to set tweak individual settings and multiple profiles.

So please, do not distract from the feature request by telling us that we can go and do it ourselves manually, because that isn't the point.

gizzygizmo, 2023-07-06:
> Having selectable speeds during a print is very handy for fine tuning (continuously variable would be best), but if you know what you want ahead of time, why not just set the speeds, accelerations, [...]

The issue is going to be novice that doesn't know the speed, accel, fan speed, etc.  They just know that for a particular tricky or vanity print they like to hit "quite" and to spit out a quick test print they may want "sport".   It's using an overal print speed adjustment and not a specific profile.   The issue with using these speed adjustments is that you can only do these while printing, and can't "save" them to your project.

Of course creating new profiles with "silent" or "ludicrious" settings for each layer height is probably the best solution here but it wasn't done for the user, there is only a "strength" and "standard" setting for each layer height.  There isn't one based on speed.

My suggestion would be to store the speed setting and/or just create more profiles that have different speed settings and make them avail to the user.  The latter is an easy fix to add, the former would require dev work but be more elegant for the end user.

aliasmeum, 2023-10-24:
Before version 1.7 it was possible to change to silent mode as soon as the slicer jumped to the device page. Now you have to wait (I believe, until the bed is heated?). If you try to change the speed before this, it just jumps back to 100%. Having to sit around waiting for the bed to heat to be able to change the speed is ridiculously annoying. Could ability to choose speed in the "Send to printer" window be an option? Preferably by setting the percentage.

Somerandomgithugguy, 2023-12-14:
> Dear BL devs,
> 
> Please add the following features:
> 
> 1. 25% and 75% speed profiles
> 2. A user-definable speed of 0-100%.
> 3. The ability to define the above "quick speed" profile for the print BEFORE printing (currently we can only do this after the print has already started).
> 
> Many thanks.

Dear BL,

Still no progress on this, REALLY? How hard is to program additional percentage markers for speeds? Or add the ability to select speed before printing?

Somerandomgithugguy, 2024-01-02:
It is so sad that this is not implemented yet. :(

kurzer90, 2024-12-04:
+1 for that!

Gask44, 2026-09-05:
+1 for a 75% speed profile.

I ran into exactly this situation today while printing a large multi-part ASA project on a Bambu Lab H2S.

At 100% I noticed the corners of some large parts beginning to warp. Switching to Silent mode at 50% helped stabilise the print, but it also increased the remaining print time dramatically.

For ASA, ABS, PC and other engineering filaments, a 75% mode would be an extremely useful middle ground. Sometimes you don't need to cut the speed in half — you just need to reduce speed and acceleration enough to give a difficult print a little more stability.

Being able to select a custom percentage such as 75% during an active print would be even better.

The current jump from 50% directly to 100% is simply too large for situations like this.