bambulab/BambuStudio #1291: Control speed slider becomes frozen on top of all applications

Somerandomgithugguy, 2023-02-17:
**Describe the bug**
When clicking the +quick speed" bar to change from 100% to 50% this causes the bar to remains frozen on-screen on top of all other applications and it will only disappear when the Bambu Studio application is closed.

**To Reproduce**
Steps to reproduce the behaviour:
1. Start a print
2. Go to 'Control' tab and select speed slider and change from 100% to 50%.

**Desktop (please complete the following information):**
 - Windows 10
 - Latest version
 - Bambu Studio 4.2.13

lanewei120, 2023-02-21:
@Somerandomgithugguy 
Hello

thanks for reporting this issue

is it a 100% reproducible issue or not?
could you give a picture of it?

Somerandomgithugguy, 2023-02-21:
Hi Lane Wei,

Yes, it is 100% repeatable and I just this moment did it on demand for you. I repeat it by starting a print and then, while the print is processing, I click the speed slider and change from 100% to 50%. Sometimes it freezes immediately on the first click, or sometimes it takes 2-3 repetitions of clicking between 100% a and 50% and then it freezes. 100% repeatable on my system. The only way to get rid of it is to close the Bambu Studio app using the X button within the application, then it disappears. I am on version 1.4.2.13 and I already tried reinstalling and it still happens.

Screenshots attached showing the speed bar frozen over all desktop apps:

Regards,

Random
.
[image]
![Bambufreeze2](https://us
[image]
er-images.githubusercontent.com/125633615/220411584-7ebf94ac-c4fb-432a-abb9-df7449b9cb61.png)

walterwongbbl, 2023-04-06:
Hi.
We will fix this issue in the next release version.

Somerandomgithugguy, 2023-04-06:
Great thanks! :D

walterwongbbl, 2023-08-02:
Please use the latest version of Bambu Studio

closed 2023-08-02 by walterwongbbl