bambulab/BambuStudio #1295: Bambu Slicer constantly Crashes

candreacchio, 2023-02-18:
Bambu slicer constantly crashes, especially on larger models. Internal version 1.4.2.13 is the version I have built. version 1.4.1.52 is much more stable.

This is under Kubuntu 22.10

If a 3MF is required please let me know.

lanewei120, 2023-02-21:
> 

Hello,

thanks for reporting this isssue

when did this crash happen?
and is it 100% reproducible

candreacchio, 2023-02-25:
Just tried with the last git pull and it still is the same issue.

See attacehd for a 3mf file
[ArmadaShipTray_v2.0.zip](https://github.com/bambulab/BambuStudio/files/10830462/ArmadaShipTray_v2.0.zip)

candreacchio, 2023-02-25:
Seems to be with multiple plates it has issues

lanewei120, 2023-02-25:
> Seems to be with multiple plates it has issues

when did this crash happen?
doing slicing or anything else

I tried to slice all the plates in your 3mf on my ubuntu
it works fine

candreacchio, 2023-02-25:
It crashes when I open the 3mf file on my machine. Sometimes if I go into
prepare first... Orbit the base plate.. then back to home and open it it
will work, but if I do nothing and try to open it crashes.

I'm using the amdgpu drivers... Is there anything I can do to help diagnose
this?

lanewei120, 2023-02-25:
Could you compile it using the debug mode "-DCMAKE_BUILD_TYPE=Debug"
and use gdb to run it to check the crash stack?

you can swith to the master newest codes, and build it following the guide
https://github.com/bambulab/BambuStudio/wiki/Linux-Compile-Guide

thanks

candreacchio, 2023-02-26:
Hmmm I built again using BuildLinux.sh -bsi instead of -dsi and it seems like its fixed? Unsure....

In anycase if i open up the app image, it does throw out a bunch of these warnings below
[log.txt](https://github.com/bambulab/BambuStudio/files/10832870/log.txt)

lanewei120, 2023-02-27:
Glad to see it works
please ignore these warnings from gtk

lanewei120, 2023-03-01:
already in v1.5 just released

closed 2023-03-01 by lanewei120