bambulab/BambuStudio #1298: Slicer does not recognize added parts, prints the same area twice

icerabbitje, 2023-02-18:
**Describe the bug**
The slicer double prints when parts are added together. 
The slicer does not recognize small added parts to combat warping. 

**3mf File for This Bug**
Any file.

**To Reproduce**
Import a large object prone to having warping corners or create a large square
Import single layer round disc to add mouse ears / corner tabs, or create single layer cylinder
Position 4 of them appropriately to the corners
Slice and review or print.
The printer will double print where there is overlap

**Expected behavior**
A) Anti Warping Tabs / Mouse Ears
Keep the main object as is, with the mouse ears /tabs attached & added around it, no double printing. 

B) Creating a larger object from different CAD parts.
Slice the objects appropriately to become one integrated model, not multiple units that may double print where intentional overlap was created.  

**Screenshots**
[image]

**Desktop (please complete the following information):**
 - OS: [e.g. iOS, Windows] Mac OS Ventura 
 - Version [e.g. 22] 13.1 / Latest

icerabbitje, 2023-02-18:
[image]

Double Printing of Square Corner and Linear Rail within the added round tab area.

Merlwynd, 2023-02-18:
You probably thought of this already…but in the meantime as a work around….export plate as stl file and reimport stl back onto plate…should combine the object appropriately?

zviratko, 2023-02-21:
I have used the "Assemble" function in the past to add a block as a pedestal to a figure. Whatever "overlapped" didn't print twice.
I also wonder how you managed to slice it at all, in my experience the slicer outright refuses to slice colliding objects ¯\_(ツ)_/¯

icerabbitje, 2023-02-21:
Will try that function next time.
BS just goes ahead with slicing after a quick alert that objects may be too
close.

Ideally adding a tab function would resolve this automatically, just as
with cura.

lanewei120, 2023-02-22:
when these two objects are seperate ones
Studio will slice them one by one, and their toolpaths are considered seperate ones also, we only add the collision detection after the slicing finished.

after assemble, they will be considered as one entire object
the slicing will remove the collision inside directly.

lanewei120, 2023-03-16:
as assemble can work as what we needs, close this issue now

closed 2023-03-16 by lanewei120