bambulab/BambuStudio #1299: Lift Z during travelling

RMKolev, 2023-02-18:
There seems to be an issue when generating supports in the middle of the print. The extruder chooses the closest path to the part, ignoring the supports and can crash directly into them. To me, this happened as the supports were in the middle of the part being printed, the lid of the extruder popped off and luckily only caused an error instead of being torn from the sockets

Ideally, there needs to be a feature such as "lifting Z" axis (in this case lowering the bed) when the printer is travelling across the print. Such feature is included in prusa slicer and Cura as lift Z, Z-hop. These features are not only done when retracting as is here, but also when travelling, to reduce chances of part collision

Alternatively, maybe the "Avoid crossing walls" can take into account supports as well. I tried this option but it still crossed into the supports and not around them.

npoulos, 2023-02-21:
Z hop is available in retract settings on printer profile and filament settings overrides

RMKolev, 2023-02-21:
From the people who I talked to on the discord server and my own browsing, it is not. This is only true for Z-hop during retraction. I am talking about Z-hop during travelling, which is not the same thing. Retraction happens at a fixed time (from my observations), while the print collision happens during the travel from one part to another, where retraction does not occur.

SaltWei, 2023-02-22:
Could you share 3MF file so we can check whether the travel can hit support? 
Thanks!

RMKolev, 2023-02-22:
https://drive.google.com/file/d/1LcAf0Tn4ycBwreDEUsJiajOEjmXRvLdT/view?usp=sharing - this is the project, plate #3 - when oriented like this, there is a big support structure in the middle of the print for the first amount of layers. The extruder starts colliding after a few layers of the supports being printed

SaltWei, 2023-02-22:
First，we have checked the sliced file and travel. All travels is higher than printed supports. So should have no hit.
Second, we also printed the model and observe the printing process and havn't seen any issue.
[image]
[image]
[image]

RMKolev, 2023-02-22:
I'm not sure if there is a big difference in the plates/printer difference. I'm running a P1P (so no lidar, not sure if the difference matters) with a textured PEI plate. I can say however I've ran the setup 3 times and it has failed on my machine - the printer head definitely collides with it, there is a very prominent noise in the early layers followed by the front coming off in the later ones. I will check the travel sizes in my sliced file. Is there any documentation how I can view it - the online 3mf reader and my notepad++ is failing to open it.

SaltWei, 2023-02-22:
> To me, this happened as the supports were in the middle of the part being printed, the lid of the extruder popped off and luckily only caused an error instead of being torn from the sockets

Do you catch the moment that the lid of the extruder pops off by hitting support?

Is the model still OK? Maybe the PEI bed of P1P is not very sticky and the model collapses which cause the issue that  lid of the extruder popped off.

SaltWei, 2023-02-22:
The model only has very small area attached on bed, and PEI bed is not as sticky as cool plate of X1

SaltWei, 2023-02-22:
[image]
You can add some support at the outer side of model to help attachment.
I think this is just issue that special model is easy to be collapsed.
P1P and X1 don't have difference which may cause this issue. Only the difference of bed. X1 is cool plate bed, and P1P is PEI bed.

RMKolev, 2023-02-22:
> Do you catch the moment that the lid of the extruder pops off by hitting support?
>
> Is the model still OK? Maybe the PEI bed of P1P is not very sticky and the model collapses which cause the issue [...]

Yes, I was watching it with my own eyes when it happened because the extruder made a noise whenever it travelled from each end of the model. The model itself was very firmly placed onto the bed (I know this because after I cancelled the print I took the model off and it was very rigidly standing). The model was definitely not affected by this.

SaltWei, 2023-03-02:
Still can not be reproduced on P1P.
My studio version is 1.5.0 Public beta on github.
Please let me known if still has problem.

closed 2023-03-02 by SaltWei