bambulab/BambuStudio #1300: Purge Yes/No or Smarter Purge - stop wasting filament by default for each new print

icerabbitje, 2023-02-18:
Whether using the same side spool, or the same spool from the AMS; when I send the next print job from BS to the printer, the printer always purges. 

When there has been no unload and reload, or filament change; there should be no purge.

It wastes time and more importantly plastic, which then contributes to more plastic going to the landfill. 

Give the user the option ( just like AMS on/off etc ) in the dialog box to say purge chamber yes/no.

zviratko, 2023-02-21:
I think all that it needs is to _not_ unload the filament from AMS at the end of the print automatically. Do that on the next print if it happens to use a different material.

icerabbitje, 2023-02-21:
Maybe ... except this is not related to AMS.

Here's the behavior that I have seen.

I recently have been printing from the side spool.
There is something in Bambu Studio that makes it purge at the start of
every print, as if it is using the AMS or a filament swap was done.  Even
though I did not change spools, hit unload, etc. The printer is always told
to purge. That is undesirable behavior.

Please give users a choice in the send to print shortcut screen ( just like
AMS yes/no ) to PURGE YES / NO, and then the software and printer can
proceed accordingly.

To always purge by default is wasteful.

zviratko, 2023-02-21:
I always use the "Unload" function on the printer when not using AMS and switching filament (although I am always tempted to skip it, cut it myself and retract...), so that shouldn't be an issue? No point purging when there was no "Unload". Or maybe just add a "Load" that purges and do no purging in the beginning of print at all? 

Part of that is actually configurable, but badly, in the initial G-Code for the printer, but this means that my ask if probably not possible with the current logic. This is in the G-Code:
"M109 S250 ;set nozzle to common flush temp"

It would make sense to set it to the max of the temperature range set in the filament profile, but I guess that would cause jams if you set that too low for the filament that was there previously (like switching from nylon to PLA), so this is a compromise.

I think someone needs to refactor the whole thing, but making it smarter probably means moving this from initial g-code to printer firmware (which could know what was there previously and at what temp it was printed), and make it start at the temperature of the old filament and finish at the temperature of the new filament. Slicer simply can't do that reliably.
Maybe adding a variable like "nozzle_temperature_initial_layer_previous_filament" could work...

icerabbitje, 2023-02-21:
Ideally, the end user should not have to try to edit g-code. Bambu Slicer
or the printer should take care of it.

When people save files on the SD card, and then go to print from there, it
may make sense to purge out of an abundance of caution.  But should it?

I see print from card as: load single filament, I take care of the purge
before hand and hit print, or load file and ask for confirmation about the
AMS 4, 8, 12, 16 colors if needed.

zviratko, 2023-02-21:
Yes, that makes sense. For other printers.
Our printers are clearly aiming to be more of an appliance - did you forget your filament? It'll tell you! Did you use the wrong bed type? It'll tell you! Is it dirty? It'll tell you... and so on.
Now back to reality, we are not there yet so you are correct, hopefully Bambu team will realize it and streamline it (for us) before X2C is out and this is EOL :)

lanewei120, 2023-03-02:
already sent to our printer team
and they will optimize the behavior in the future, for details please sync it on the bambulab forum

RyanEwen, 2023-07-28:
A checkbox or dropdown option on the Print Plate dialog would be really really appreciated. 

The option to purge or not purge is nice, but I would also like the option to purge extra (in case of changing from black to white, PLA to PETG, etc). 

Current workaround is to go to the Device section, click the new color, click Load, wait, and then Print Plate. That way it purges twice. Still not quite enough to go from black to white, though.