bambulab/BambuStudio #1301: Print End Time and Number of Filament Changes left, Home Screen addons

Benderrodrigu3z, 2023-02-18:
The new feature for Layers printed is great but can we add to this?
Completion Time? 3:30pm
Number of Filament Changes left? 3/20
The Bambu Lab Image perimeter on the home screen could be slightly smaller and perhaps Customizable Color Options or at the very least show all the AMS devices or colors on that screen at the bottom

Benderrodrigu3z, 2023-02-18:
Filament changes left is usefully say if our AMS is acting up or if one filament is having issues backspinning half way through the print

chris9012, 2023-02-25:
I was just going to put in a request for the print end time.  I think this would be super helpful and is surprising most printers and slicers don't have this functionality today.

Benderrodrigu3z, 2023-02-25:
In network request for time zone Sincerely,Shane FSent from my iPhone13

Chompworks, 2023-08-12:
> Number of Filament Changes left? 3/20
This would be so useful - on both the BambuStudio progress bar, and also on the printer UI if we could get it.

GuzziRaz, 2024-01-05:
> Number of Filament Changes left? 3/20

Came here to ask for this. 👍