bambulab/BambuStudio #1302: more specific means of dynamic adjusting layer height

tsdexter, 2023-02-18:
**Is your feature request related to a problem? Please describe.**
A clear and concise description of what the problem is. Ex. I'm always frustrated when [...]

I cannot adjust the layer height starting a specific height and going up to another specific height and to a specific value. The slider makes it very hard to do this accurately and get every layer between the exact same height

**Describe the solution you'd like**
A clear and concise description of what you want to happen.

A layer height tool exactly like the "paint by height range" tool would be perfect except rather than selecting a paint color you enter the layer height, then you also enter then height range as you do with painting, then you move the mouse over the model to select where to start the height change. 

**Describe alternatives you've considered**
A clear and concise description of any alternative solutions or features you've considered.

can't think of any, the new way with the slider provides more control but it's hard to use effectively 

**Additional context**
Add any other context or screenshots about the feature request here.

MadMax389, 2023-02-22:
Agreed, this would be an awesome feature. One workaround I have used is to reset the minimum layer height in the printer profile. If I want my .2mm print to have a .16mm section, set the minimum to .16 and then use the mouse to saturate the section to the minimum. Not exact, and cludgy if you want more than 2 layer heights, but better than trying to guess with the mouse.
[image]

tsdexter, 2023-02-26:
@MadMax389  I found a good solution in the FB group, which seems obvious now, but you can use modifier parts with a specific thickness and starting z-height and then adjust their layer height, however, you don't get the "smooth transition" this way

MadMax389, 2023-02-26:
Great idea, I'll use that in the future.