bambulab/BambuStudio #1303: P1P Blocked from using PLA Silk profile

TheJoyofPrinting, 2023-02-19:
In bambu studio if you do not have the X1C selected in printer settings menu you cannot see or select the PLA Silk profile. even after turning on the X1C profile you cannot use the silk profile if you have the P1P selected as the printer. As soon as you switch to the X1C you can use the PLA Silk profile.

1. Make sure you only have the P1P selected
2. try to select the pla silk profile
3. now select the X1C printer as well
4. Look for and select pla silk profile
5. now in the main view of the slicer have the P1P selected 
6. tick the drop down menu on any of the filament drop downs
7. now select the X1C as the printer 
8. repeat step 6

**Expected behavior**
I should be able to select and use the PLA Silk profile with the P1P selected as my printer

[image]
[image]

 - OS: [windows 11]
 - Version [1.4.2.13]

SaltWei, 2023-02-21:
PLA silk profile is missing for P1P, will add back

TheJoyofPrinting, 2023-02-21:
Thank you very much!

SaltWei, 2023-03-02:
Available in V1.5.0 Public beta

closed 2023-03-02 by SaltWei