bambulab/BambuStudio #1305: not storing data in Appdata (Windows)

b-reich, 2023-02-19:
**Is your feature request related to a problem? Please describe.**
BambuStudio need Files in `%appdata%`, this makes provding an package a bit more difficult. 

**Describe the solution you'd like**
It would be need, if there is an portable version of BambuStudio. 

**Describe alternatives you've considered**
Is it possible to create an "portable" Installer?
Or helping me to fullfil my PR for scoop? https://github.com/ScoopInstaller/Extras/pull/10175

julie777, 2023-02-22:
When I go to File/Help/Show Configuration Folder it opens %USER_PROFILE%\AppData\Roaming\Bambu Studio which has all the configuration files.

Can you clarify your comment?

b-reich, 2023-02-23:
I have seen there are drivers for the webcam integration of the printers. Without this, the webcam does not work. At least that is my observation when I extract the installer and start the slicer.

closed 2025-05-11 by b-reich