bambulab/BambuStudio #1307: Black not black in AMS menu causing auto spool change to fail

3dprintandpaint, 2023-02-19:
**Describe the bug**
AMS colour choice on the printer menu isn't black, although it appears to be.  When synced in Studio, the colour is actually dark grey.  This causes issues when a spool ends and you expect it to move across to the same filament on another AMS slot, which may have been assigned black in Studio.  

**3mf File for This Bug**
N/A

**To Reproduce**
Steps to reproduce the behaviour:
1. Select Black via the AMS option on the printer menu.
2. Go to Bambu Studio and sync the AMS with studio
3. Colour shown is very dark grey, not black
4. Filament option used to replicate - Bambu PLA basic

**Expected behaviour**
Choosing black on the screen should be black in the slicer.

**Screenshots**
N/A

**Desktop (please complete the following information):**
Windows 11

bambu123, 2023-03-02:
[image]

Are you mean this black color? It displays such as on my computer.

[image]

bambu123, 2023-03-02:
It is really RGB #161616.

bambu123, 2023-03-02:
#1170

3dprintandpaint, 2023-03-02:
Yes, that's the colour 👍

[image]

bambu123, 2023-03-02:
I think it should be modified on the Printer to use the real black color.
I shall send the issue to the printer team.

bambu123, 2023-03-02:
@3dprintandpaint
What is the color of your new filament? 
Did you assign the color to it from the touch screen of the printer?

3dprintandpaint, 2023-03-02:
I put a roll of black filament into the AMS and assign it black via the touch screen.  When syncing in Studio, it isn't black it is grey.  The colour on the printer screen may well represent grey, which if it does, a black colour option is necessary next to the grey.

bambu123, 2023-03-08:
AMS auto spool change requires the same colors between the new and old trays, If you assign them the same color, then it should work.
I posted the issue to our printer team, who may answer the problem.

3dprintandpaint, 2023-03-08:
I think the original message is getting lost and I'm not sure how else the phrase this.  I am fully aware of how the AMS works. 

It may be better if I rephrase this -

Please ADD a black to dot colour option to the AMS menu.  When you press this BLACK dot, the spool colour is black, not grey.

bambu123, 2023-03-08:
I think you are requesting more color options on the Printer touchscreen and Bambu Handy App, right?

3dprintandpaint, 2023-03-08:
I guess if the current dot, which I assumed was black, is in fact grey, then yes there needs to be a black option.  Currently, because of the dot size, it can easily be mistaken as black when in fact it's grey.  Black is a very popular colour and yet it is missing from the menu.

Adding black to the menu will negate the risk of people thinking (like me and #1170) that the current grey dot is black.

bambu123, 2023-04-23:
Fixed in version 1.6, please have a try.

Fixed by add AMS filament color for selecting.

closed 2023-04-23 by bambu123