bambulab/BambuStudio #1310: Missing Layers in Vase Mode

laisch-at, 2023-02-19:
When Slicing the added Model in Vase Mode, I always get missing layers at about 23.8/ 54.0 and 84mm

Looks like the missing layers are always on the height where the inside has the same shape.

G-Code and STL provided in the ZIP
[wing_test_files.zip](https://github.com/bambulab/BambuStudio/files/10777488/wing_test_files.zip)

Bambu Studio V 1.4.2.13

SaltWei, 2023-02-21:
This is special 3D printed airplane model and should not use vase mode directly. This is not a bug.
Should use settings as following.
[image]
[image]

closed 2023-02-21 by SaltWei

laisch-at, 2023-02-21:
Your provided settings are working fine thx for that.

But my primary intention when making this wing design for testing, was to print it in vase mode, like I did with Cura and my older printer.
In Cura this technique worked totally fine on airplane wing models I designed myself using this technique 
Attached screenshots for comparison: On Bambuslicer you see the shift in Z - direction.

[image]
[image]

SaltWei, 2023-02-22:
This is not a bug, just setting difference. If you still want to use vase mode for this model, please set this setting to be 0 as well.
[image]
BUT, for more other airplane model, we recomand using even-odd mode and settings as I provided in previous comment.

andershedberg, 2024-04-14:
I also thing there is a bug in the vase-mode. Some things are simply missing in the preview and later prints, and the top and bottom layer are not done at all so the holes for rods are completely missing. I got the impression that the vase mode was to keep the printhead moving in a constant speed to avoid stringing and "jumps", but here it sort of misses other things and part of the wing profile... (I used the vase mode settings from the bambu wiki page, and I think it affected that model as well as it had no "top" cover, but maybe it should have...)

P3skyBee, 2024-11-04:
> This is not a bug, just setting difference. If you still want to use vase mode for this model, please set this setting to be 0 as well. [image] BUT, for more other airplane model, we recomand using even-odd mode and settings as I provided in previous comment.

Wanted to let you know that this has fixed the same issue for me after days of struggling with everything I could think of to fix it. Thank you so much