bambulab/BambuStudio #1313: If Download directory is renamed, Bambu Studio will not start

TheyWho, 2023-02-20:
**Describe the bug**
Bambu Studio [1.4.2.13] SETTINGS saves a default location for DOWNLOADS.
If this download directory is renamed or deleted [outside of the program], then the next time you try to start Bambu Studio, it will show an error that the path cannot be found and exit out.  The ONLY solution is to recreate the missing path.

**3mf File for This Bug**
If it is related to slicing, please append the 3mf file. It could be extremely helpful to solve the issue.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to 'Preferences'
2. Go to 'Downloads'
3. Click on [Browse]
4. Set to a temporary directory
5. Exit Bambu Studio
6. Rename the directory or any directory higher in the path
7. Try to restart Bambu Studio Ver 1.4.2.13
8. Nope, you can't get in. (Windows 10).

**Expected behavior**
I would expect the program to start and allow you to choose another directory.

**Desktop (please complete the following information):**
 - OS: Windows 10
 - Version 1.4.2.13
 - 
**Screenshots**
[image]
[image]

SaltWei, 2023-02-22:
@walterwongbbl Please check this issue

walterwongbbl, 2023-02-23:
Hi.
we will fix this issue in the next release version.

lanewei120, 2023-03-03:
already fixed in v1.5

joseRey, 2023-06-11:
I had the same issue just now.  I created a new directory and moved the file to it and Babu Studio will not re-start.

lanewei120, 2023-06-12:
> I had the same issue just now. I created a new directory and moved the file to it and Babu Studio will not re-start.

@joseRey 
could you provide the version info and logs just as described in 
https://github.com/bambulab/BambuStudio/issues/new?assignees=&labels=bug&projects=&template=bug_report.yml

walterwongbbl, 2023-06-30:
Hi.
Could you please provide detailed step-by-step instructions?

walterwongbbl, 2023-10-13:
Please download and use the latest version of Studio

closed 2023-10-13 by walterwongbbl