bambulab/BambuStudio #1314: Accidentally clicking on X or Y movement during print crashes print head into sides and needs hard restart

Fl0ken, 2023-02-20:
**Describe the bug**
When accidentally clicking the buttons to move the print head in x or y direction, the print head moves around the chamber and crashes into the sides. Only a hard reset fixes this issue, stopping the print does nothing. Issues like this can happen when you want to change the speed or the fans in BambuStudio

**To Reproduce**
Steps to reproduce the behavior:
1. Start a print
2. Click on any direction of the x or y axis while printing
3. See error

**Expected behavior**
The printer should ignore x or y input during the print or at least lock the controls which then have to be manually unlocked.

**Desktop (please complete the following information):**
 - OS: OSX
 - Version: any, currently 1.4.2.

LiZ-Li-BBL, 2023-02-23:
Sorry we can't reproduce this problem. What is your OS and version? Is the button disabled(show as grey) or normal when you click it? It will be nice to have a picture description.

StoneLiBambulab, 2023-10-24:
Close this issue as fixed, open a new one if there is any problem.

Thanks!

Stone Li

closed 2023-10-24 by StoneLiBambulab