bambulab/BambuStudio #1315: Allow only one wall on top surfaces together with Arachne perimeter generator

liftbag, 2023-02-20:
Personally I find the "Only one wall on top surfaces" feature almost indispensable for improving the finish of 3D printed objects, especially those with large horizontal surfaces.
Coming from SuperSlicer, I really appreciated the presence of this feature also on Bambu Studio (unlike PrusaSlicer, which continues to ignore user requests in this regard).
However, I realized that by enabling Arachne, the single perimeter on the tops can no longer be enabled.
So, if the slicing engine allows it, why not allow the two parameters to coexist, as is possible on SuperSlicer?

randyre, 2023-02-21:
Would also like to see one wall on the bottom layer, fo any mode not just Arachne. Sometimes the bottom layer is the layer that once removed is the top of the object.

SaltWei, 2023-02-22:
> So, if the slicing engine allows it, why not allow the two parameters to coexist, as is possible on SuperSlicer?

Accept and add into develop plan. Thank you very much for suggestions!

SaltWei, 2023-02-22:
> Sometimes the bottom layer is the layer that once removed is the top of the object.

How about a setting one wall on first layer?

PinGuiNPL, 2023-02-22:
I've been wanting this for a long, long time... Initially, SuperSlicer also turned off the option for single wall top surfaces when you activated Arachne.
I don't know how it is now because I've only been working with SF-BambuSlicer for several months.
One walled top surface with Arachne would be SUPER!

liftbag, 2023-02-23:
> ... Initially, SuperSlicer also turned off the option for single wall top surfaces when you activated Arachne.

It may be that the problem with SuperSlicer was never a problem at all.
Currently, including the latest nightly builds, if Arachne is enabled in SuperSlicer, the option to activate the single perimeter on top layers is greyed out.
But if you choose the classic perimeter generator, tick the Only one perimeter on top surfaces and come back to Arachne, the option is fully functional.
This does not happen with Bambu Studio, nor with its fork.
For this reason I am confident that the two options can easily coexist.

DzzD, 2023-04-07:
+++++ for both bottom and top aswell as arachne walls

fti7, 2023-05-05:
+1 for Bottom :-(

QingZhangBambu, 2023-08-08:
only one wall for the first layer and only one wall for Arachne had be implemented, you can get full functionality software after we release 1.7 hotfix2.

closed 2023-10-16 by QingZhangBambu