bambulab/BambuStudio #1316: TPU taking forever to slice and stalling out

Benderrodrigu3z, 2023-02-21:
[double padding.STL.zip](https://github.com/bambulab/BambuStudio/files/10788958/double.padding.STL.zip)

here is a file from a website, its free but It can only be printed in TPU, well the slicer just sits there forever on 1.04.02.13 at 65% at layer 149
Macbook Catalina 10.15.7

I sliced another item in TPU and it took forever as well but what's more interesting is that this is a new thing to happen because I opened 01.03.01.0.1 and its much faster at slicing it but I loose a lot of the newer features. Please fix this.

If I click on cancel or do anything in the slicer it goes to Not Responding, so far its taken over an hour to get to this point and I don't know if anything is even happening

Benderrodrigu3z, 2023-02-21:
Material: TPU 85A
Brand name: Ninjatek - Ninjaflex
Nozzle: 0.4mm
Layer height: 0.15mm
Infill: 25%
Top, bottom layers: 4
[image]

Wall layers: 2

Benderrodrigu3z, 2023-02-21:
the issue happens from Tree Support branch diameter being set to 1, I set it to 2 and it compiles but set to 1 has no issues on the older version

SaltWei, 2023-02-22:
@tangketan Bug about tree support. Please have a check. Thanks!

tangketan, 2023-03-10:
The support has nothing to do with material. And I tried on the latest version with branch diameter 1mm with no problem. However, I do encounter problems with branch distance set to 1. This is because there are too many branches to grow, and this is universal for many models. I'll try to solve this problem.

tangketan, 2024-03-19:
Problem solved. Will be seen in the next release.