bambulab/BambuStudio #1319: [bug] LAN mode not working on MacOS

psiberfunk, 2023-02-22:
As of the latest firmware update and slicer I am unable to use LAN mode with macOS with the printer. On the same network with the same printer I am able to use LAN mode with a Windows machine , but I cannot even get to the prompt asking for the LAN security code from the printer on macOS.. it just hangs when trying to connect .

 I have tried this from two macos machines just to make sure .. they are both running ×86 versions of the slicer.

I’ve tried reinstalling on macOS , and resetting the printer to factory , and toggling LAN mode , nothing seems to work .

StoneLiBambulab, 2023-03-08:
Hi @psiberfunk ,
please try the 1.5.0 beta version https://github.com/bambulab/BambuStudio/releases/tag/v01.05.00.59.
and if lan mode still not working, could you help to upload your bambu studio log for us?
please refs to https://wiki.bambulab.com/en/x1/troubleshooting/bambu-studio-log

Thanks!

Stone Li

brianredbeard, 2023-03-12:
Attempted with version 1.5.0.60 

Bambu Studio detects the printer over the network, prompts for the access code, and then fails:

```
Connect 3DP-00M-000 failed! [SN:00M00AAAAAAAAAA, code=-1]
```

Unfortunately as 1.5.0.60 still fails to connect I'm forced me to revert back to 1.4.0.18.

psiberfunk, 2023-03-12:
Also still failing.

brianredbeard, 2023-03-14:
Issue persists in 1.05.0.61.

psiberfunk, 2023-03-14:
Can confirm, issues still here.

StoneLiBambulab, 2023-04-06:
Hi @psiberfunk,
[image]

Please try to delele the config files (BambuStudio.conf, BambuNetworkEngine.conf) and reinstall the bambu studio. 
Step:  Menu-> Help -> Show Configuration Folder 

Thanks!
Stone Li

spodec, 2024-08-29:
Hi,

I am still having the same issue. Deleting the config files and reinstalling Bambu Studio didn't solve the problem. Is there any other solution meanwhile?

Thanks.

[image]

raechten, 2026-09-12:
I had this issue. I found that MacOS never registers that BambuStudio is allowed Local Network access (you can check this in System Settings -> Privacy and Security -> Local Network. You will not see BambuStudio here (if you do and it is off, turn it on and it should work). To get it to work, I first turned on Local Network for all devices on the network using the Terminal, using the following command (only do this if you feel comfortable doing this.  Make sure the IP range matches yours. Should be in the form of X.X.X.0): 

sudo defaults write com.apple.network.local-network AllowedWiFiLocalNetworkAddresses -array "192.168.0.0/24"

Then reboot your Max (this is essential).

Afterwards you can open BambuStudio and should be able to add your printer using IP/Access Code if it doesn't show up by itself (mine didn't). 

If this works, you can remove the systemwide exception by using the following command in Terminal:

sudo defaults delete com.apple.network.local-network AllowedWiFiLocalNetworkAddresses

Reboot again (IMPORTANT!!!)

Afterwards, you should see BambuStudio in the Local Network list (in System Settings -> Security & Privacy). If it is OFF, turn it on. When starting BambuStudio it should now show your device(s). At least this worked for me. 

It seems to be an issue with MacOS's network security. I've seen reports that it is fixed in the new MacOS 27 versions but I haven't checked this.