bambulab/BambuStudio #1320: Split by Part + X-Y Hole Compensation causes jagged edges

ctadlock, 2023-02-22:
If a multi-solid model is imported and it is split by part and "X-Y hole compensation" is non-zero, the resulting slicing has small holes in it. This does nopt happen if the model is split by object or if the xy hole compensation is 0. 

This is not model related and can be reproduced with and multi-solid model.

**To Reproduce**
Steps to reproduce the behavior:
1. Import a multi-solid model (use attached as an example)
2. Set "X-Y hole compensation" to 0.1
3. Split by part
4. Slicer

[box.zip](https://github.com/bambulab/BambuStudio/files/10801408/box.zip)

**Expected behavior**
The slicing is clean.

- Windows 11
- 1.4.2.13
- Also happens with Softfever

**Screenshots**

Difference between split by object and part (two examples)
[image]

[image]

Poor quality result
[image]

High quality result when split by object (two examples)
[image]
[image]

ctadlock, 2023-02-22:
SuperSlicer has this same issue. I think everyone just copied the same code/bug:

[image]

QingZhangBambu, 2023-04-03:
the color object should better not use X-Y hole compensation, we will add a warning msg

QingZhangBambu, 2023-04-21:
we will fix it as fast as possible

QingZhangBambu, 2023-06-13:
you can set the resolution to 0.0 which could fix that

closed 2023-06-13 by QingZhangBambu

ctadlock, 2025-03-27:
> you can set the resolution to 0.0 which could fix that

This did not fix the issue

[image]

ctadlock, 2025-03-28:
@QingZhangBambu this should be reopened