bambulab/BambuStudio #1323: Send to printer SD card-Files can not be renamed

aliasmeum, 2023-02-22:
https://github.com/bambulab/BambuStudio/issues/1183#issue-1558254009

I am afraid this is not solved, and that it is more serious than the “Rename file while sending” function not working. When using 3mf-files from Fusion360 it seems the name of the original part in Fusion360 is stuck no matter what you call the file. When you change filename, or change something in a project, and then send it to the printer you have no way of knowing what file is actually on the SD-card because it names the file the same as before the change. This behavior leaves the “Send to printer” function untrustworthy and thereby useless in the current state. I am really happy that Bambulab added this functionality, since the “Send to printer” function is excellent to use for files that are printed repeatedly, and I hope they fix this as soon as possible.

This is what I have done to try to reproduce the issue.

Fusion360:
-Save “A v1” as mesh in Fusion > “A.3mf”
-Save “B v2” as mesh in Fusion > “B.3mf”

Windows10 File explorer:
-Rename file “A.3mf” > “A-Fusion.3mf”
-Rename file “B.3mf” > “B-Fusion.3mf”

Bambu Studio 1.4.2.13:
-New project
-Add > “A-Fusion.3mf”
-Save project as > “A-Studio.3mf”
-Slice
-Send
Result: File sent to printer is named “A v1”

[image]

[image]

[image]

-New project
-Add > “B-Fusion.3mf”
-Save project as > “B-Studio.3mf”
-Slice
-Send
Result: File sent to printer is named “B v2”

[image]

[image]

[image]

-Open project > “A-Studio.3mf”
-Delete “A-Fusion.3mf”
-Add “B-Fusion.3mf”
-Save project as > “AtoB-Studio.3mf”
-Slice
-Send
Result: File sent to printer is named “B v2”

[image]

[image]

[image]

[image]

Merlwynd, 2023-02-22:
Just wondering, have you tried closing project, then reopening with new project name in BS before sending to printer?

aliasmeum, 2023-02-22:
Same result.

sodachen22, 2023-02-24:
@aliasmeum The printer shows project name instead of file name in this window. Did you set the name of the project? Could you please share A-Fusion.3mf?

aliasmeum, 2023-02-24:
Hi @sodachen22
If you look at the photos from my last example you can see that the printer is not using the project name. You can see that the project name is "AtoB-Studio" on the next photo you can see that BambuStudio confirms the file export of "AtoB-Studio_plate_1.gcode.3mf" and on the last photo you can see that the file is named "B v2" on the printer screen. "B v2" is the name of the design in Fusion360 and this is the name BambuStudio uses when exporting to the printer. It does not matter if you change the name when you save as mesh in Fusion360, renames the file in Windows file explorer or change the project name in BambuStudio. It seems like BambuStudio exports the file with the name of the first file you open in a project. And if you renamed the file when saving as mesh in Fusion360 BambuStudio is not using this new filename, but the original name of the design in Fusion360. I find it a bit difficult to explain this, but I hope it still makes sense.
Files are attached:

[AB.zip](https://github.com/bambulab/BambuStudio/files/10823888/AB.zip)

sodachen22, 2023-03-01:
The name on the X1 screen is the model name which can be changed in the project tab. Looks like the design confused the user, we will consider the way for optimization.
[image]

aliasmeum, 2023-03-01:
Thank you @sodachen22
I can confirm that you are right, the model name under the project tab is the name you see on the printer. And you are also right that the design confused the user, this is not a behavior I have ever seen in any software I have used. In my normal process from opening a file to sending it to the printer, BambuStudio uses at least 5 different file names. I will try to explain:

-Name 1 (A-Fusion): I create a new project and open a 3mf-file from Fusion360 that I have named "A-Fusion". As expected BambuStudio uses this name in the top of the window.

-Name 2 (A v1): In the project tab, model name field, BambuStudio uses the name of the design in Fusion360. This is not the name I used when saving the file as mesh in Fusion (A-Fusion), but the name of the Fusion design file (A v1).
Expected behavior: The name of the first file you open in a project should automatically be transferred to the model name field (A-Fusion). BambuStudio should NOT EVER use the design file name from Fusion (A v1).
[image]

-Name 3 (A-Project): I change the name in the model name field from "A v1" to "A-Project" and return to the prepare tab. Now I expect that I have opened a design file into a project, renamed the project and I then want to save it. This is when I get really concerned. When I save the project, BambuStudio overwrites the design file (A-Fusion). So now I have lost my design file from Fusion called "A-Fusion" and in stead I have a project file from BambuStudio with the same name.
Expected behavior: When saving the project for the first time, the save as window should open! The name I just chose in the project tab (A-Project) should automatically be transferred to the file name field in the save as window. If I choose to save the project with another name, the model name should automatically be changed to the same name.
[image]

-Name 4 (A-Fusion_plate_1): I slice the model and press the sent to printer button. BambuStudio now adds "_plate_1"
 to the filename and I have file name no. 4. The rename function in this window is not doing anything, since no matter what you rename the file it will be transferred to the printer with the name in the model field under the project tab.
Expected behavior: The rename function in this window should override anything else and the name you create here should be the name on the file you see on the printer.
[image]

-Name 5 (A-Fusion_plate_1.gcode.3mf): After pressing send in the sent to printer window BambuStudio confirms the export of the file to the printer and adds ".gcode" to the name. This new name only exists here in this confirmation popup window, since the name on the transferred file always will be the name in the model field under the project tab.
Expected behavior: This file name should be a confirmation of the file name I used in the send to printer window and should be the name of the file after transferring to the printer.
[image]

My suggestion:
1-When a project is saved for the first time, the save as window must be used.
2-The model name field in the project tab should not be a editable field, but should reflect the file name. If you rename the file in for example windows file editor or use the save as function to save the project from BambuStudio the model name field should be updated to show the new name.
3-The name on the file that you can see on the printer should always be the name that is chosen in the send to printer window.

aliasmeum, 2023-03-14:
This still applies to the new version 1.5.0.60
When you save a project for the first time, the save as window is not used and BambuStudio overwrites the first file you added to the project. This is a serious problem that must be fixed.

aliasmeum, 2023-05-18:
Thanks to Bambulab for continuing to make great improvements to the printer firmware and to Bambu Studio. The experience of using the printer keeps getting better and better.

But, in version 1.6.2.4 there is still problems with naming and saving.
-The most serious issue is that when you save a project for the first time, the save as window is not used and BambuStudio overwrites the first file you added to the project. This means that you original file is lost and not recoverable.
-The send to printer function is still not naming properly. If you open a previously saved project, make some changes and use the save as function to save the project with a new name, the send to printer function keeps using the old project name.

walterwongbbl, 2024-01-18:
Please use the latest version of BambuStudio.
Version 1.8 or higher is required.

closed 2024-01-18 by walterwongbbl

aliasmeum, 2024-01-29:
This is not solved in version 1.8.4.51
-The most serious issue is that when you save a project for the first time, the save as window is not used and BambuStudio overwrites the first file you added to the project. This means that you original file is lost and not recoverable. This is not expected behavior.