bambulab/BambuStudio #1324: Better Highlighting Of Selected Object On The Object Tab In Dark Mode

MadMax389, 2023-02-22:
**Is your feature request related to a problem? Please describe.**
In a large project with multiple plates (in dark mode), it is very hard to see which object is selected in the object list. The background color of the selected object is only slightly different than the dialog background. (It's easily visible if not in dark mode).
[image]

**Describe the solution you'd like**
Change the background of the selected object to one that works in dark mode and light mode, green or red, for example. Optionally, change the text color.

**Describe alternatives you've considered**
None, really. Hunt down the part on the plate.

**Additional context**
[image]

bambu123, 2023-03-02:
Mac or Windows?

MadMax389, 2023-03-02:
Windows 11. My apologies, I should have have specified in the original post.

bambu123, 2023-03-09:
I think it is the background color of the selected item should be changed.
Text color is static in both dark and light modes and does not need to be modified.

psy0rz, 2023-04-25:
check out https://github.com/bambulab/BambuStudio/releases/tag/v01.06.00.65

can probably be closed now.

bambu123, 2023-04-26:
Yes, it's fixed in version 1.6, though not pretty.

closed 2023-04-26 by bambu123