bambulab/BambuStudio #1325: Macro level view of a Print Farm style setup

PaulM3D, 2023-02-22:
I'm increasing the number of Bambu machines rapidly as we replace older, slower machines, but seeing an overview of what is going on across them all is tedious and more of a problem due to the speed of the machines (more swap outs needed more frequently).  All machines have an AMS and are a mix of P1P and X1 series.

I'd like to see as a first itteration a dashboard style view that allows a view of machine status and AMS filaments on each machine to help identify a machine to send a job to knowing it has the filament available and is either finished or almost finished.  Even better would be to be able to send jobs from multiple Studio installs (4 users in the office for example) into a single queue and the jobs can then be allocated to a machine as they become available.

PaulM3D, 2023-04-05:
No thoughts or feedback on this?  We'll be at 20 machines soon!

StoneLiBambulab, 2023-04-05:
Hi @PaulM3D ,
      We want to do this and plan for it, but we do not have enouph developers to do it now. There's so much to do.

Thanks!
Stone Li