bambulab/BambuStudio #1327: Provide Feedback to User When Cloud is Unavailable

rehashedsalt, 2023-02-23:
**Is your feature request related to a problem? Please describe.**
Cloud service availability is opaque and, during outages like today's, prints sent to the printer fail silently.

**Describe the solution you'd like**
One of the following should happen in the user interface of Bambu Studio when a BambuLab printer is in the device list and the slicer cannot connect to Bambu Cloud:
* The existing "Failed to Connect to Network" banner should appear; or
* When a print is sent to the printer and fails, an error message should appear; or
* The device should be set as disconnected

**Additional context**
This issue is being filed during an outage of Bambu cloud printing services, thus the motivation.

zviratko, 2023-02-23:
Moreover, I think this should behave like an atomic transaction

1) slicer sends the print to cloud
2) slicer sends a message to the printer to fetch it and start printing (or so I understand it)
2) the printer downloads the file and start printing

this is insufficient, as sending the print doesn't guarantee the print actually starting, and even worse, in case of outage, you might send several prints and the printer just picks one of them seemingly randomly

I propose it should work like this
1) slicer sends the file to the cloud
2) slicer sends a message for the printer to commence printing
3) printer downloads and validates the file
4) printer sends message back to the slicer that everything is cool and it's about to print
5) slicer ackowledges and displays "Actually going to print, I promise!"
6) printer receives the ack and starts printing

I have no idea how it works technically, maybe MQTT is used and supports something like this, but I think this is needed to avoid random prints starting during/after an outage, or even starting another print right after one has finished during outages which has catastrophic results.

icerabbitje, 2023-02-26:
The behavior and functionality that I expect when the cloud is not available is, is that there is a prompt:
cloud networking is not available, would like to send this directly to the printer instead? 
Yes/no. 
Yes. Establish a direct connection via WiFi. Done.