bambulab/BambuStudio #1334: [*Issue* Bambu Studio Crashes when loading STL] Bambu Studio crashing upon loading a STL File

BambuLabDiscord, 2023-02-24:
Trying to load a specific STL file and bambu studio crashes with no error message. Happens everytime on a specific STL file.

lanewei120, 2023-02-25:
could you share the stl file to us for a debug?

TopNerdJR, 2023-03-04:
https://www.thingiverse.com/thing:5889636

Uploaded the stl file to thingiverse. Ive also tried loading the same file as a 3mf and it starts trying to load the file into the slicer then automatically closes Bambu Studio. Ive also updated to a beta version to test (Both 1.4 and 1.5) and I still get the same result. File is created in fusion360

lanewei120, 2024-06-26:
already fixed, can load it successfully now
[image]

closed 2024-06-26 by lanewei120