bambulab/BambuStudio #1336: Markers and arrows size

suchabadperson, 2023-02-25:
The behavior of markers and arrows in Bambu Slicer is not ideal, especially on high resolution screens.

In PrusaSlicer markers are larger, and they grow or shrink as you move closer or further away from the object.
On the contrary, in BambuSlicer they always have the same size, quite small and sometimes it makes difficult to select or modify the object.

[image]

kertgartner, 2024-03-27:
Thank you! Hopefully they'll fix it.

Haidiye00, 2024-10-16:
The new version 1.10beta has been released. Please download and use~

closed 2024-10-16 by Haidiye00